jgromes/RadioLib #3: Runtime Transmit error

juliantech5, 2018-09-29:
Hey Jgromes, 
I seem to be having a runtime error when I try to use the Kitelibe Library with RTTY instead of LoRaLib. It is the same exact piece of code and the only difference is the library and the Lora module declaration

`#include <SoftwareSerial.h>
#include <KiteLib.h>
#include <TinyGPS.h>
 
TinyGPS gps;
SoftwareSerial ss(6, 5);
SX1278 lora = new Module(7, 3, 2);
 
void setup()
{
  Serial.begin(115200);
  ss.begin(9600);
 
 int  state = lora.begin(434.0, 62.5, 12, 8, 0x13, 10);
  if (state == ERR_NONE)
  {
    Serial.println(F("success!"));
  }
  else
  {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}
 
void loop()
{
  bool newData = false;
  unsigned long chars;
  unsigned short sentences, failed;
 
  for (unsigned long start = millis(); millis() - start < 1000;)
  {
    while (ss.available())
    {
      char c = ss.read();
      if (gps.encode(c))
        newData = true;
    }
  }
 
  if (newData)
  {
    float flat, flon;
    unsigned long age;
    gps.f_get_position(&flat, &flon, &age);
    Serial.print("LAT=");
    Serial.print(flat == TinyGPS::GPS_INVALID_F_ANGLE ? 0.00 : flat, 7);
    Serial.print(" LON=");
    Serial.print(flon == TinyGPS::GPS_INVALID_F_ANGLE ? 0.00 : flon, 6);
    Serial.print(" ALT=");
    Serial.print(gps.f_altitude() == TinyGPS::GPS_INVALID_F_ALTITUDE ? 0.00 : gps.f_altitude());
 
    String transmissionMessage = String("");
   
    transmissionMessage += String(";");
    if (0==1)
    {
      transmissionMessage += String("0.0");
    }
    else
    {
      transmissionMessage += String(flat, 4);
    }
 
    transmissionMessage += String(";");
    if (0==1)
    {
      transmissionMessage += String("0.0");
    }
    else
    {
      transmissionMessage += String(flon,4);
    }
 
    transmissionMessage += String(";");
    if (0==1)
    {
      transmissionMessage += String("0.0");
    }
    else
    {
      transmissionMessage += String(gps.f_altitude());
    }
 
    int state = lora.transmit(transmissionMessage);
   if (state == ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(" success!");

    // print measured data rate
    Serial.print("Datarate:\t");
    Serial.print(lora.dataRate);
    Serial.println(" bps");

  } else if (state == ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 256 bytes
    Serial.println(" too long!");

  } else if (state == ERR_TX_TIMEOUT) {
    // timeout occured while transmitting packet
    Serial.println(" timeout!");

 
  }
 }
}`
This code shown above transmits once and then locks up the loop.

`#include <SoftwareSerial.h>
#include <KiteLib.h>
#include <TinyGPS.h>
 
TinyGPS gps;
SoftwareSerial ss(6, 5);
SX1278 lora = new Module(7, 3, 2);
 RTTYClient rtty(&lora);
void setup()
{
  Serial.begin(115200);
  ss.begin(9600);
 
 int  state = lora.begin(434.0, 62.5, 12, 8, 0x13, 10);
  if (state == ERR_NONE)
  {
    Serial.println(F("success!"));
  }
  else
  {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}
 
void loop()
{
  bool newData = false;
  unsigned long chars;
  unsigned short sentences, failed;
 
  for (unsigned long start = millis(); millis() - start < 1000;)
  {
    while (ss.available())
    {
      char c = ss.read();
      if (gps.encode(c))
        newData = true;
    }
  }
 
  if (newData)
  {
    float flat, flon;
    unsigned long age;
    gps.f_get_position(&flat, &flon, &age);
    Serial.print("LAT=");
    Serial.print(flat == TinyGPS::GPS_INVALID_F_ANGLE ? 0.00 : flat, 7);
    Serial.print(" LON=");
    Serial.print(flon == TinyGPS::GPS_INVALID_F_ANGLE ? 0.00 : flon, 6);
    Serial.print(" ALT=");
    Serial.print(gps.f_altitude() == TinyGPS::GPS_INVALID_F_ALTITUDE ? 0.00 : gps.f_altitude());
 
    String transmissionMessage = String("");
   
    transmissionMessage += String(";");
    if (0==1)
    {
      transmissionMessage += String("0.0");
    }
    else
    {
      transmissionMessage += String(flat, 4);
    }
 
    transmissionMessage += String(";");
    if (0==1)
    {
      transmissionMessage += String("0.0");
    }
    else
    {
      transmissionMessage += String(flon,4);
    }
 
    transmissionMessage += String(";");
    if (0==1)
    {
      transmissionMessage += String("0.0");
    }
    else
    {
      transmissionMessage += String(gps.f_altitude());
    }
 
    int state = lora.transmit(transmissionMessage);
   if (state == ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(" success!");

    // print measured data rate
    Serial.print("Datarate:\t");
    Serial.print(lora.dataRate);
    Serial.println(" bps");

  } else if (state == ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 256 bytes
    Serial.println(" too long!");

  } else if (state == ERR_TX_TIMEOUT) {
    // timeout occured while transmitting packet
    Serial.println(" timeout!");

     rtty.begin(434, 183, 45, 5);
   
    rtty.println(transmissionMessage);

  lora.standby();

  Serial.println(F("done!"));

  // wait for a second before transmitting again
  delay(1000);
  }
 }
}`

This is ultimately what I am trying to achieve but it gets locked after the LoRa transmission.  

Thank you very much Jgromes, hope I'm not disturbing too much with these questions :)

Best Regards,
Julian

jgromes, 2018-09-30:
I was unable to reproduce this issue; the first code snippet repeatedly transmits fine for me.

Couple of ideas for debugging:
1) DO NOT under any circumstances use Arduino String class. It can easily lead to heap fragmentation and crash. Use C-strings instead.
2) Your DIO0 and DIO1 are swapped. Are you sure the code corresponds to your hardware?
3) In the second code, you're starting the LoRa module twice. Also, for RTTY to transmit anything, you must use FSK modem.

closed 2018-09-30 by jgromes

juliantech5, 2018-09-30:
Hey, 
Thanks for your reply, I am not sure If it is hardware related or software but it gets stuck exactly at Lora transmit. The RTTY and Lora get decoded successfully but then it goes quiet without printing "Done!"
https://pastebin.com/UbrgpZT4

juliantech5, 2018-09-30:
Also on a side note, Using the default sketch with ITA2, Numbers get send incorrectly

jgromes, 2018-09-30:
ITA2 data type expects C-strings, not numbers. You need to convert your numbers to string and then to ITA2.

jgromes, 2018-09-30:
Still cannot replicate your issue with the pastebin code; everything transmits correctly.

jgromes, 2018-09-30:
Got it.

>Your DIO0 and DIO1 are swapped. Are you sure the code corresponds to your hardware?

They are. When I swap them in my code, it stops because the TxDone interrupt is never received. It will probably time out after a few minutes.