jgromes/RadioLib #4: RTTY on CC1101?

yasiupl, 2018-11-13:
Hi! I'm looking for a solution to transmit RTTY teletype telemetry through a CC1101 equipped shield. I see that you're working on something that might satisfy my needs. What's the outlook and is there maybe an ETA?

Also, is KiteShield needed to use your library?

jgromes, 2018-11-13:
Hi, KiteLib doesn't support CC1101 at the moment. I wasn't planning to add it, so I don't even have the module - even if I ordered it right now, it would take some time to arrive.

Also, do you know if it is possible to get RTTY working on CC1101? The way it is implemented for SX127x and RFM69 chips is a bit of a hack. I haven't used CC1101 before, so I don't know if it will work. Is there a specific reason you want to use that module?

KiteShield isn't required to use the library, you can connect as many different wireless modules as you like - or at least until your Arduino runs out of memory ;)

jgromes, 2018-11-13:
If you want to get the maximum distance, why not something like LoRa? It was used in high-altitude baloons before (e.g. https://github.com/LoRaTracker/HAB2). LoRa chips like SX1278 are pretty common and cheap.

jgromes, 2018-11-13:
Still not a reason not use the SX1278 chips though :P Apart from LoRa, they can also transmit FSK (and have RTTY support here), so if you don't want to use LoRa modulation you don't have to. FSK on SX1278 can go up to 300 kbps. If it was me designing the system, I would choose the part that gives me the most options.

Also, I'd say these chips are pretty hackable, I managed to broadcast a .wav file with Arduino, SD card and SX1278 - I doubt that anyone who designed them ever thought of this "use case" ;)

yasiupl, 2018-11-13:
Thanks for the insight! I will look into it more and check what's within my budget.

jgromes, 2018-11-13:
OK, I'll close this for now, feel free to reopen later.

jgromes, 2018-12-19:
@yasiupl just got some CC1101 modules, are you still interested in this?

jgromes, 2018-12-24:
I think I know how http://canduino.eu managed to get RTTY working on CC1000 - they're using it as a simple 2-FSK radio, and set the frequency deviation to 300 Hz, so it ends up looking like RTTY with 300 Hz shift between mark and space frequency.

The problem with CC1101 is that it can't set frequency deviation that low - the lowest value I can set to the configuration registers is about 1.6 kHz. You also have to keep in mind that is the shift from _carrier_, so shift between mark and space will be 3.2 kHz in this case. I don't know whether you will be able to receive that as RTTY (I'm using SDR dongle with SDR Console v2, and I can only go up to 850 Hz).

As I've said before, the way RTTY is implemented on SX127x/RF69 is a massive hack - basically, I just turn on the transmitter and shift the carrier frequency. On SX127x chips, that can be set in 61 Hz steps (with 32 MHz crystal), which is good enough for RTTY with 183 Hz shift. 

On CC1101 however, that step size is 397 Hz (26 MHz crystal). That's better than the 3.2 kHz shift, but still quite a lot. So my questions is this - if I manage to get RTTY working, will you be able to receive it with your hardware?

jgromes, 2018-12-26:
Basic RTTY support on CC1101 is in now, so you can try it. I set the frequency shift to 397 Hz (the lowest available value) and I can receive RTTY fine with RTL-SDR dongle and SDR console v2 (RTTY shift set to 425 Hz).

Please note that none of the packet funcions are  working on CC1101, it's just RTTY for now.

yasiupl, 2018-12-26:
Cool, looks promising, I will check it out asap.

jgromes, 2019-01-09:
Closed due to inactivity, feel free to reopen later.

closed 2019-01-09 by jgromes