jgromes/RadioLib #5: Newbie starting point - module connection or pin definitions ?

mr-sneezy, 2019-01-15:
Where do you have the pin connection info for non KiteShield hook ups ?
I'm trying to use the RFM69_receive example, with an Arduino UNO pinout compatible board (a Wemos D1) and a RFM69 breakout board to suit (but not your shield). 
What Arduino pins go to the SPI pins on the RFM69 module for your code examples ?
I looked over all the files but came up blank...

jgromes, 2019-01-15:
I know the documentation is pretty much non-existent at this point - sorry about that, I'm working on a new version of the shield, proper docs are the next item on the TODO list ;)

You can use custom pinouts by specifying them to the `Module` constructor, like this:
```c++
RF69 rf = new Module(cs, dio0, dio1);
```

mr-sneezy, 2019-01-15:
Ah yeah, OK that's still above my understanding (a module constructor). 
What would be gold is a simple drawing of your connection from the module pins in your shield and the Arduino pins (or a PDF circuit diagram of the KiteShield would be enough too).
Cheers,
Martin

jgromes, 2019-01-15:
You can use the RF69 example, only replace the line 

```c++
RF69 rf = Kite.ModuleA;
```

with 

```c++
RF69 rf = new Module(cs, dio0, dio1);
```

where cs, dio0 and dio1 are the pins you want to use for SPI chip select, RF69 DIO0 and DIO1.
It's also using the standard SPI pins (MISO, MOSI, CLK), although I'm not sure where those are on Wemos D1.

mr-sneezy, 2019-01-15:
Hmm. 
Looking at my breakout. I have an Elecrow RFM69 shield like this [https://www.elecrow.com/wiki/index.php?title=RFM69_Shield-_434MHZ](url)

Looking at the schematic for that and my Wemos, if I plug it as a shield the pin connections are:
RFM69 - Wemos D1 Arduino:
DPIO0 - D0
SCK     - D5 (SCK)
MISO  - D6 (MISO)
MOSI  - D7 (MOSI)
These are via a HC4050 level shifter but I left that out as it not relevant to the end to end points.
Is that info enough to create the required the code line you show above ?

jgromes, 2019-01-15:
Looking at the Eagle file, DPIO0 is connected to what would be Arduino pin D2 and CS (NSS) to Arduino pin D10 (I don't know if those numbers correspond to wemos pinout),

DPIO1 is not wired on the shield you linked. In my opinion, that's a major design oversight, since RF69 reports receive timeouts on that pin, so each time you use `receive()` method, it will end up in an infiinite loop (until a packet is received). Alternatively, you could use the interrupt-driven `startReceive()` and `readData()` methods - see the RF69_Receive_Interrupt example for details (that one only uses DPIO0). Although I don't know if D2 on wemos supports external interrupts (on Arduino Uno it does).

Your constructor will look like this (on Arduino):
```c++
RF69 rf = new Module(2, 10, 3);
```
SPI CS is D2, DPIO0 is D10 and DPIO1 is unconnected, so I just left the default D3 in there (you have to specify all 3 pins).

jgromes, 2019-01-15:
On a side note, I'm pretty sure wemos D1 is a 3.3V board, and the shield has a 5V logic shifter, so you will be applying 5V voltage to a 3.3V board - probably not a good idea.

mr-sneezy, 2019-01-15:
The Wemos D1 does have different Arduino IO pin numbering to the UNO. [https://arduino.stackexchange.com/questions/49119/arduino-uno-r3-to-wemos-d1-r2-project-migration-pinout-problems](url)

I think I should just drop the Wemos D1 for now and breadboard the shield to a Nano to get the example running (I don't have an Uno on hand) and leave the shifter in place. I seem to be making it harder to progress by using the Wemos D1 right now :)
Thanks for your info BTW, a little goes a long way at this stage in my 'boot strapping'.