jgromes/RadioLib #9: SX126x support

jgromes, 2019-04-08:
Add SX1262/68 support to RadioLib.

Main issues:
1. SPI interface is different than SX127x series.
2. Additional BUSY pin is required.

Originated in https://github.com/jgromes/LoRaLib/issues/53

addaedans, 2019-04-10:
Hi! Sorry for the late reply! Here is the link where i found the files:
http://www.dorji.com/products-detail.php?ProId=65

At the left side there is the Resources tab and there is the reference code. Hope it helps to get it working :)

jgromes, 2019-04-10:
Thanks, it looks like a slightly modified SX1262 driver from https://github.com/Lora-net/LoRaMac-node

jgromes, 2019-05-18:
SX126x support is implemented as of f0b9841cf69a7b8551b99c28b3b50c2bbfb1d1e9, it should be at the same level as SX127x support now.

closed 2019-05-18 by jgromes