jgromes/RadioLib #18: RFM95 aka SX127x not detected

beegee-tokyo, 2019-06-04:
Using [SparkFun LoRa Gateway - 1-Channel (ESP32)](https://www.sparkfun.com/products/15006)
NSS = 16, DIO0 = 26, DIO1 = 33, DIO2 = 32 according to [schematic](https://cdn.sparkfun.com/assets/5/3/1/9/c/ESP32_LoRa_1_Channel_Gateway.pdf)
Setting up the connection to the RFM95 module:
```cpp
RFM95 lora = new Module(16, 26, 33, 32);

void setup() {
    int state = lora.begin(868.0, 125.0, 9, 5, 0x3444, 22, 8);
    if (state == ERR_NONE)
    {
        Serial.println("success!");
    }
    else
    {
        Serial.print("failed: ");
        printError(state);
        return false;
    }
...
}
```
fails with
```log
=====================================
RFM95 sender test
=====================================
DeviceID: 84 0D 8E 0C 56 3C 00 00
Initialize RFM95 
SX127x not found! (1 of 10 tries) SX127X_REG_VERSION == 0x00, expected 0x0011
SX127x not found! (2 of 10 tries) SX127X_REG_VERSION == 0x00, expected 0x0011
SX127x not found! (3 of 10 tries) SX127X_REG_VERSION == 0x00, expected 0x0011
SX127x not found! (4 of 10 tries) SX127X_REG_VERSION == 0x00, expected 0x0011
SX127x not found! (5 of 10 tries) SX127X_REG_VERSION == 0x00, expected 0x0011
SX127x not found! (6 of 10 tries) SX127X_REG_VERSION == 0x00, expected 0x0011
SX127x not found! (7 of 10 tries) SX127X_REG_VERSION == 0x00, expected 0x0011
SX127x not found! (8 of 10 tries) SX127X_REG_VERSION == 0x00, expected 0x0011
SX127x not found! (9 of 10 tries) SX127X_REG_VERSION == 0x00, expected 0x0011
SX127x not found! (10 of 10 tries) SX127X_REG_VERSION == 0x00, expected 0x0011
No SX127x found!
failed: SX127x not found
```
Using the abandoned library [RadioHead](https://github.com/PaulStoffregen/RadioHead)
Setup with
```cpp
// Singleton instance of the radio driver
RH_RF95 rf95(16, 26);
// 16 = NSS, 26 = INT

void setup() {
  // Init the RFM95 module
  if (!rf95.init())
  {
    Serial.println("[RFM95] init failed");
    delay(500);
  }
}
```
works perfect, log output:
```log
RH_RF95::init() started
RH_RF95::init() success
```

Any idea?

jgromes, 2019-06-04:
During initialization, RadioLib checks the version register of RFM95. It should read 0x11, yours returns 0x00. That usually indicates faulty SPI bus or damaged device. It looks like RadioHead performs no such check, does the device actually work?

beegee-tokyo, 2019-06-04:
Yes the device works perfectly. The counterpart is [SparkFun Pro RF - LoRa, 915MHz (SAMD21)](https://www.sparkfun.com/products/14916) and the two communicate without problems with each other.

But......

Found the problem.
I am using an ESP32 which has two SPI, one called HSPI and a second one VSPI

The RF96 is connected to HSPI, but it seems the ESP32 defaults SPI to VSPI.
So the SPI is initialized on the wrong pins. That's why it doesn't work. 
But no idea why it works with the RadioHead library.

As a quick test I hard-coded the pins into Module::init()
```cpp
      _spi->end();
      _spi->begin(14, 12, 13, 16);
```

Then it works. 
Now I need to figure out how to set the SPI to the correct pins and give it as argument to 
```
RFM96 lora = new Module(16, 26, 33, 32, mySPI);
```

Sorry for bothering you with this. It's not a bug, but an ESP32 specific problem.

closed 2019-06-04 by beegee-tokyo