jgromes/RadioLib #20: Semtech DRF1278DM - Connection UART

Benjamincreator, 2019-06-08:
Hi there,

 Trying to work out how to use a Semtech DRF1278DM [http://www.dorji.com/products-detail.php?ProId=34](url) with an Arduino Nano, I've read the other posts on here regarding how LoRaLib won't do UART and only SPI but the KiteLib link just redirects to the main LoRaLib page! 

So if the LoRaLib library now supports the DRF1278DM module and not just the raw SX1278 board, how would you go about connecting it up, as it doesn't have the standard SCK and MOSI? 
 
Would it just be like this? 
`SX1278 lora = new LoRa(11, 10);`

Cheers Benjamin!

jgromes, 2019-06-08:
As is explained in LoRaLib [contributing doc](https://github.com/jgromes/LoRaLib/blob/master/CONTRIBUTING.md), that library is intedned as a driver for bare SX127x radios. The one you have has a UART interface.

This library (RadioLib) was based on LoRaLib, and extended to cover all sorts of wireless modules. However, I don't plan on adding support for DRF1278DM, for the following reasons:
1. Some users could be confused over what interface to use with SX127x - SPI or UART.
2. DRF1278DM isn't the only UART SX1278-based module out there, with different incompatible UART interfaces (e.g. RN2903).
3. The UART interface only offers a very limited set of functions, when compared to bare SX1278. (e.g. no CAD, no interrupts, no FSK modem etc.)
4. Most people use the bare SX1278.

closed 2019-06-08 by jgromes

Benjamincreator, 2019-06-08:
Okay understood! Cheers!