jgromes/RadioLib #21: SX126x - Set DIO3 as output

beegee-tokyo, 2019-06-14:
I have another weird module, a combined nRF52832 and SX1262 module named [Insight ISP4520](https://www.insightsip.com/products/combo-smart-modules/isp4520). 
This module uses DIO3 as a power supply for the antenna switch:
[image]

They have a sample code how to set DIO3 as output and put it to high level:
```cpp
static void SX126xDio3Control (bool state)
{
    uint8_t reg_0x0580;
    uint8_t reg_0x0583;
    uint8_t reg_0x0584;
    uint8_t reg_0x0585;
    uint8_t reg_0x0920;

    if (state)
    {
        // Configure DIO3 as output

        // Read 0x0580
        SX126xWaitOnBusy();
        GpioWrite(&SX126x.Spi.Nss, 0);
        SpiInOut(&SX126x.Spi, RADIO_READ_REGISTER);
        SpiInOut(&SX126x.Spi, (0x0580 & 0xFF00) >> 8);
        SpiInOut(&SX126x.Spi, 0x0580 & 0x00FF);
        SpiInOut(&SX126x.Spi, 0);
        reg_0x0580 = SpiInOut(&SX126x.Spi, 0);
        GpioWrite(&SX126x.Spi.Nss, 1);

        // Read 0x0583
        SX126xWaitOnBusy();
        GpioWrite(&SX126x.Spi.Nss, 0);
        SpiInOut(&SX126x.Spi, RADIO_READ_REGISTER);
        SpiInOut(&SX126x.Spi, (0x0583 & 0xFF00) >> 8);
        SpiInOut(&SX126x.Spi, 0x0583 & 0x00FF);
        SpiInOut(&SX126x.Spi, 0);
        reg_0x0583 = SpiInOut(&SX126x.Spi, 0);
        GpioWrite(&SX126x.Spi.Nss, 1);

        // Read 0x0584
        SX126xWaitOnBusy();
        GpioWrite(&SX126x.Spi.Nss, 0);
        SpiInOut(&SX126x.Spi, RADIO_READ_REGISTER);
        SpiInOut(&SX126x.Spi, (0x0584 & 0xFF00) >> 8);
        SpiInOut(&SX126x.Spi, 0x0584 & 0x00FF);
        SpiInOut(&SX126x.Spi, 0);
        reg_0x0584 = SpiInOut(&SX126x.Spi, 0);
        GpioWrite(&SX126x.Spi.Nss, 1);

        // Read 0x0585
        SX126xWaitOnBusy();
        GpioWrite(&SX126x.Spi.Nss, 0);
        SpiInOut(&SX126x.Spi, RADIO_READ_REGISTER);
        SpiInOut(&SX126x.Spi, (0x0585 & 0xFF00) >> 8);
        SpiInOut(&SX126x.Spi, 0x0585 & 0x00FF);
        SpiInOut(&SX126x.Spi, 0);
        reg_0x0585 = SpiInOut(&SX126x.Spi, 0);
        GpioWrite(&SX126x.Spi.Nss, 1);

        // Write 0x0580
        SX126xWaitOnBusy();
        GpioWrite(&SX126x.Spi.Nss, 0);
        SpiInOut(&SX126x.Spi, RADIO_WRITE_REGISTER);
        SpiInOut(&SX126x.Spi, (0x0580 & 0xFF00) >> 8);
        SpiInOut(&SX126x.Spi, 0x0580 & 0x00FF);
        SpiInOut(&SX126x.Spi, reg_0x0580 | 0x08);
        GpioWrite (&SX126x.Spi.Nss, 1);

        // Write 0x0583
        SX126xWaitOnBusy();
        GpioWrite(&SX126x.Spi.Nss, 0);
[... 273 lines cut ...]
      return state;
    }
    return ERR_NONE;
  }
  else
  {
    {
      state = readRegister(0x0920, &reg_0x0920, 1);
      if (state != ERR_NONE)
      {
        return state;
      }
      reg_0x0920 = reg_0x0920 & ~0x08;
      state = writeRegister(0x0920, &reg_0x0920, 1);
      if (state != ERR_NONE)
      {
        return state;
      }
    }
  }
  return ERR_NONE;
}
```

PS: I have no idea where Insight got the register addresses 0x0580, 0x0583, 0x0584, 0x0585 and 0x0920 from. They are not in the datasheets. I guess they got it from Semtech directly.

jgromes, 2019-06-14:
You don't have to add a new `SPItransfer` method; the one that's currently in place is the base, and all the other SPI methods should be routed through it. You can simply add the address bytes to the `dataOut` buffer. The reason `readRegister` isn't implemented was that I had no use for it.

I'm very much against adding this to the library, we have no idea what's in the registers at those addresses, and what else might be affected. Unless Semtech decides to release a complete register map, I'm afraid I can't risk this.

Side note: this is my main issue with SX126x series - the SPI interface is abhorrent. Not only are now most (not all) exchanges done through commands, the SPI registers weren't eliminated completely, and I have a sneaking suspicion that all the commands do is set the register values. Why the f*** won't they document all the SPI registers, allowing developers to access them directly - which would be much simpler than the command interface?

beegee-tokyo, 2019-06-14:
I agree with the your points regarding the undocumented registers and the weird way the SX126x SPI interface is build.
I really don't understand why they didn't make it as simple as on the SX127x chips.
I will keep it in my local version of the library, because I need it.

But regarding ```SPItransfer()```, my problem is that if I call it with ```write = false``` it writes only the ```cmd``` and then starts writing ```NOP``` while receiving the data into ```dataIn```, but I need to write 3 bytes before starting with the ```NOP```.
When I call  ```SPItransfer()``` with ```write = true``` it writes the ```cmd``` and then the bytes in from ```dataOut``` but the received bytes are thrown away and not returned in ```dataIn```.

Two ways to change this:
**a)** change ```uint8_t cmd``` to ```uint8_t *cmd, uint8_t cmdLen``` and then do
```cpp
for (uint8_t idx = 0; idx < cmdLen, idx++)
{
  spi->transfer(cmd[idx];
}
```
but that would break the code in many places.

or **b)**
```cpp
  if (write)
  {
    for (uint8_t n = 0; n < numBytes; n++)
    {
      // send byte
      uint8_t in;
      if (dataIn != NULL)
      {
        in = dataIn[n] = spi->transfer(dataOut[n]);
      }
      else
      {
        in = spi->transfer(dataOut[n]);
      }
      DEBUG_PRINT(dataOut[n], HEX);
      DEBUG_PRINT('\t');
      DEBUG_PRINT(in, HEX);
      DEBUG_PRINT('\t');

      // check status
      if (((in & 0b00001110) == SX126X_STATUS_CMD_TIMEOUT) ||
          ((in & 0b00001110) == SX126X_STATUS_CMD_INVALID) ||
          ((in & 0b00001110) == SX126X_STATUS_CMD_FAILED))
      {
        status = in;
      }
    }
    DEBUG_PRINTLN();
  }
```

jgromes, 2019-06-14:
> but that would break the code in many places.

Would it? I think it will break the code in exactly 2 places: in `SPIreadCommand` and `SPIwriteCommand`, both of which are an easy fix (just replace the `cmd` variable with 1-byte array). After that, you can just call `SPItransfer` from `readRegister`.

beegee-tokyo, 2019-06-14:
Right, I got confused when searching for calls of SPItransfer.

Anyway, this code is nothing that would go into your library, so let's just keep it here for reference.

closed 2019-06-14 by beegee-tokyo

ericlangel, 2025-03-04:
is the ISP4520 Module actual supported by this library, or not?