jgromes/RadioLib #22: How can I get the raw data on DIO2 with interrupt ?

AjJordy, 2019-07-03:
**IMPORTANT: Check the wiki**
Before submitting new issue, please check the [Wiki](https://github.com/jgromes/RadioLib/wiki) and the [API documentation](https://jgromes.github.io/RadioLib/). You might find a solution to your issue there. 

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

jgromes, 2019-07-03:
Reading the raw data is not supported by the library and I'm not planning on adding it in the near future. Reasoning behind that is this feature probably wouldn't be used by many people.
 
The only thing receiveDirect() does is it puts the radio into direct reception mode, so that DIO1 and DIO2 act as serial clock and data output. I have never checked the output to be honest - I never had a use for the serial data stream.

closed 2019-07-03 by jgromes

giordand, 2020-07-12:
@AjJordy , could you find any solution to this? I'm trying to do exactly the same thing. I have an sx1278 and a 434 mhz radio control

AjJordy, 2020-07-13:
I'm using this [https://github.com/rrobinet/RFM69X_Library](Repo)

AjJordy, 2020-07-13:
I'm using this [https://github.com/rrobinet/RFM69X_Library](repo)

giordand, 2020-07-13:
Ok, but I'm using a Mini SX1278 LoRa Transceiver Module

AjJordy, 2020-07-13:
Looking for something like Continuous Mode, Promiscuous Mode. In my case, I needed to configure by SPI each register.

giordand, 2020-07-13:
Yes, I did that and I can see that raw data changes when i press the button of my control, but i don't now how to save that signal in order to send it again