jgromes/RadioLib #23: fatal error: SoftwareSerial.h: No such file or directory

richard-ubi, 2019-07-12:
I just didn't see Software Serial. h in the project. I guess it was uploaded or something. I compiled it using Arduino platform. Look forward to it. Thank you.

jgromes, 2019-07-12:
SoftwareSerial is used for modules with UART interface like ESP8266 or HC-05. It's a standard Arduino library, so if your Arduino installation is missing it, you have to reinstall.

closed 2019-07-12 by jgromes

richard-ubi, 2019-07-12:
Thank you very much for your reply. I use ESP32. I guess besides the library, I need to transplant and modify the code inside.

Richard@ubitech.hk
 
From: Jan Gromeš
Date: 2019-07-12 17:05
To: jgromes/RadioLib
CC: richard-ubi; Author
Subject: Re: [jgromes/RadioLib] fatal error: SoftwareSerial.h: No such file or directory (#23)
SoftwareSerial is used for modules with UART interface like ESP8266 or HC-05. It's a standard Arduino library, so if your Arduino installation is missing it, you have to reinstall.

jgromes, 2019-07-12:
Sorry, this library wasn't intended to work on ESP8266/ESP32. The only actively supported platform is AVR (though I've seen it run on others).

richard-ubi, 2019-07-12:
Well, let me try to modify your lib to make esp32 available.

Richard@ubitech.hk
 
From: Jan Gromeš
Date: 2019-07-12 17:30
To: jgromes/RadioLib
CC: richard-ubi; Author
Subject: Re: [jgromes/RadioLib] fatal error: SoftwareSerial.h: No such file or directory (#23)
Sorry, this library wasn't intended to work on ESP8266/ESP32. The only actively supported platform is AVR (though I've seen it run on others).

jgromes, 2019-07-12:
Sure, why not. But please take a carfeul look at [CONTRIBUTING.md](https://github.com/jgromes/RadioLib/blob/master/CONTRIBUTING.md).

richard-ubi, 2019-07-19:
Thank you very much！I have successfully used ESP32 to drive SX1261 from your library. Thank you .

Richard@ubitech.hk
 
From: Jan Gromeš
Date: 2019-07-12 18:13
To: jgromes/RadioLib
CC: richard-ubi; Author
Subject: Re: [jgromes/RadioLib] fatal error: SoftwareSerial.h: No such file or directory (#23)
Sure, why not. But please take a carfeul look at CONTRIBUTING.md.

jgromes, 2019-07-19:
That's great! Could you share the code by opening a pull request, so that we can add ESP32 support for everyone? Thanks.

richard-ubi, 2019-07-19:
sorry! i can't understand how to do it ! but i can send my lib to u !Actually, it hasn't changed much, just blocked the development of arduino.

Richard@ubitech.hk
 
From: Jan Gromeš
Date: 2019-07-19 12:43
To: jgromes/RadioLib
CC: richard-ubi; Author
Subject: Re: [jgromes/RadioLib] fatal error: SoftwareSerial.h: No such file or directory (#23)
That's great! Could you share the code by opening a pull request, so that we can add ESP32 support for everyone? Thanks.

jgromes, 2019-07-19:
Opening a pull request is pretty simple - you have to fork this repository to your GitHub account, make the changes you want in that fork and open the pull request, which will allow me to merge the changed forked into this repository.

>blocked the development of arduino

I'm not entirely sure what you mean by that - could you elaborate?