jgromes/RadioLib #24: scanChannel without timeout in while, infinty loop.

Sallatiel, 2019-07-16:
Hi, 
I am using arduino pro micro - atmega32u4 (INT0 and INT1 are pins exchanged in relation to arduino uno) and in scanChannel the code gets stuck in while waiting for an activity in INT0. In the comment says have timeout but do not have, should not have a timout? I am only with a radio and now I have no way to test if with channel activity it would pass this condition.

thank you! and congratulations to cubeSat

jgromes, 2019-07-16:
CAD timeout is controlled by the radio module itself, because CAD length depends on LoRa configuration. If you end up in an infinite loop it suggests incorrect wiring.
1. What radio module are you using?
2. Does it have TCXO? And if so, did you set the reference voltage?
3. What's the wiring?

jgromes, 2019-07-30:
Closed due to inactivity.

closed 2019-07-30 by jgromes

Sallatiel, 2019-08-05:
Hello,

1. EBYTE E22-900M22S;
2. Yes, I saw that it has TCXO in DIO3 but I don't know what value to set in setTCXO;
3. In Arduino Pro Micro (ATmega32u4):
Sx1262 - Arduino Pro Micro (ATmega32u4)
NSS ----------- 10 (PB6)
SCK ----------- 15 (PB1)
MOSI --------- 16 (PB2)
MISO --------- 14 (PB3)
BUSY --------- 0 (PB5)
DIO1 ---------- 3 (PD1 - INT0)
DIO2 ---------- 2 (PD0 - INT1)

jgromes, 2019-08-05:
If your module is using TCXO, you have to configure it by calling `setTCXO()`, otherwise it won't work. Apparently, 2.4 works with E22-900M22S (see https://github.com/jgromes/RadioLib/issues/12#issuecomment-497321332), but you should confirm that with the seller.

Sallatiel, 2019-08-05:
Thanks!

working. I didn't find the crystal datasheet, is it a TCXO or VCTCXO? Is this pin on DIO3 crystal VDD or control? Thank you.

[image]

jgromes, 2019-08-05:
It's most likely a TCXO - see SX1262 datasheet, section 4.1.4. DIO3 provides the supply voltage from 1.6 to 3.3 V. Then again, I'm only guessing here - you should probably contact the seller or the manufacturer if you want more details.