jgromes/RadioLib #26: RTTY in RFM69

fazerlab, 2019-07-28:
I couldn´t use RTTY with RFM69 Module. 
From the example code (https://github.com/jgromes/RadioLib/blob/master/examples/RTTY/RTTY_Transmit/RTTY_Transmit.ino ) i change to:

RF69 fsk = new Module(10,2 ,3 );
----------------------------------------------------------------------------------
get error in compilation:
radiolib_rtty:46: error: 'class RF69' has no member named 'beginFSK'
int state = fsk.beginFSK();
exit status 1
'class RF69' has no member named 'beginFSK'
---------------------------------------------------------------------------------------------
Documentation say: 
RTTY for modules: SX127x, RFM9x, SX126x, RF69, SX1231, CC1101 and nRF24L01

I´m doing something wrong?

jgromes, 2019-07-28:
Try `begin()` instead of `beginFSK()` - only LoRa modules have `beginFSK()` method, because they need to distinguish between LoRa and FSK modem.

fazerlab, 2019-07-29:
Thank you. I got the code compiled with begin() instead of beginFSK()

closed 2019-07-29 by jgromes