jgromes/RadioLib #27: Compile Error using RadioLib.h

alexisvieiro, 2019-07-29:
Hi, i am using a sx1272 module (ra-01) with a esp12f as microcontroller.
I try to compile a simple code but the ARDUINO IDE give me this error:

> command-line>:0:9: error: expected identifier before numeric constant
> /home/alexis/Arduino/libraries/RadioLib/src/modules/ESP8266.h:13:7: note: in expansion of macro 'ESP8266'
>  class ESP8266: public TransportLayer {
>        ^
> command-line>:0:9: error: expected unqualified-id before numeric constant
> /home/alexis/Arduino/libraries/RadioLib/src/modules/ESP8266.h:13:7: note: in expansion of macro 'ESP8266'
>  class ESP8266: public TransportLayer {
>        ^
> exit status 1
> Error compilando para la tarjeta Generic ESP8266 Module.

Code: (reduced for simplicity)

> #include <RadioLib.h>
>   
> void setup() {
> }
> 
> void loop() {
> }

Any ideas? Thank you

jgromes, 2019-07-29:
ESP8266 is not supported, because one part of RadioLib is a UART driver for ESP8266 with AT firmware, so there's a conflict in names (`ESP8266` class from RadioLib vs. a macro named `ESP8266` defined by the platform).

I can add a few macro checks to exclude RadioLib's ESP8266 from compilation on ESP8266 platform. However, you obviously won't be able to use the `ESP8266` driver. All other modules will work just as on any AVR Arduino.

alexisvieiro, 2019-07-31:
Ok, i added  "#ifndef  ESP8266" and "#endif" in the ESP8266.c and ESP8266.h files and its works without errors. Thank you.

closed 2019-07-31 by alexisvieiro