jgromes/RadioLib #33: Issue with Interrupts using SX1262MB2CAS Shield from Semtech

cvoalex, 2019-08-06:
Hi

I can't for the life of me get the Transmit or Interrupt examples working.   Seems the library is not firing the interrupt.

This is what I am using:
https://www.mouser.com/ProductDetail/Semtech/SX1262MB2CAS?qs=qSfuJ%252Bfl%2Fd69tTGSeW1ezg%3D%3D&gclid=CjwKCAjwyqTqBRAyEiwA8K_4O7C7a8jZO-cE0LqnMlDPHCW88X2zbjaKHPTFOQHc3fD0QoWU1JnpBhoCh3AQAvD_BwE

The regular examples work fine.
I am on an Arduino Uno and the initialization I am using is as follows:
SX1262 lora = new Module(7, 5, -1, 3);

 int state = lora.begin(915.0, 125.0, 9, 5, SX126X_SYNC_WORD_PRIVATE, 22);

Any help will be greatly appreciated.

jgromes, 2019-08-06:
You're assigning DIO1 to Arduino pin 5 - Arduino Uno only has two external interrupts on pins 2 and 3, so you have to use those if you want interrupts.

closed 2019-08-06 by jgromes

cvoalex, 2019-08-06:
Thank you so much for the prompt reply

All the best