jgromes/RadioLib #34: Recieve with interrupt on SX1268 without using DIO2

rnbokade, 2019-08-12:
I want to use recieve with interrupt on SX1268 however I am constrained by the no. Of pins I have available. So can I just put DIO2 pin as -1 and still use the recieve interrupt feature?

jgromes, 2019-08-13:
I'm not sure I understand the question - you always need to connect DIO2 to some Arduino pin in order to receive data, regardless of whether it's blocking or intrrupt-driven reception.

rnbokade, 2019-08-13:
Ok so connecting DIO2 to some pin is necessary?? So is there some pin which I could ditch without compromising my ability to send Lora packet, Send FSK packet, receive FSK packet with interrupt.
My application is very low power one thus using SX1268 series is essential for it and also I have used up almost all pins so need to ditch one of the pins of SX1268. Is it possible to use your library without one of the pins.

jgromes, 2019-08-13:
Sorry, I just noticed the question was about SX126x, not SX127x. On SX126x, you can omit DIO2 and still have access to all the features. So yes, you can set DIO2 as - 1.

rnbokade, 2019-08-13:
Thank you very much for such instant replies. I and my team are very thankful that you took time to reply to us, solved our big-time problem. Also thank you very much for developing such a god-level library.

closed 2019-08-13 by rnbokade