jgromes/RadioLib #35: Support for MSK,GMSK.

rnbokade, 2019-08-15:
Docs claim that Semetech chips have capabilities for MSK and GMSK. Can those be invoked using some settings in code?

jgromes, 2019-08-17:
You didn't mention whether you mean SX127x or SX126x series, but the answer is pretty much the same for both: What's the source of that information? (G)MSK is only mentioned in passing, and as far as I can tell, there's no way to actually use it.

rnbokade, 2019-08-18:
https://www.semtech.com/products/wireless-rf/lora-transceivers/sx1268
Check in the features section.

jgromes, 2019-08-18:
Features section isn't docs - it's advertising. Judging by the actual docs (i.e. datasheet), SX126x either can't do MSK at all, or it is completely undocumented.

closed 2019-08-18 by jgromes

rnbokade, 2020-03-26:
Minimum frequency-shift keying or minimum-shift keying (MSK) is a particular spectrally efficient form of coherent FSK. In MSK, the difference between the higher and lower frequency is identical to half the bit rate. Consequently, the waveforms that represent a 0 and a 1 bit differ by exactly half a carrier period. The maximum frequency deviation is δ = 0.25 fm, where fm is the maximum modulating frequency. As a result, the modulation index m is 0.5. This is the smallest FSK modulation index that can be chosen such that the waveforms for 0 and 1 are orthogonal.
Source :- Wikipedia

rnbokade, 2020-03-26:
So I guess a function beginMSK could be added which in turn can call befinFSK to set the deviation 0.25 times that of bitrate... Later the shaping filter value is already known to set gaussian filter... So GMSk is also solved

jgromes, 2020-03-26:
Is there a reason to add it then? Since MSK is just a special case of FSK, I don't see any advantage in implementing a new `beginMSK` method - it would likely confuse users more than anything.

rnbokade, 2020-03-26:
At least one can add note somewhere in the examples or wiki, regarding this.

jgromes, 2020-03-26:
Again - I'm not sure it's really beneficial. Users that need MSK in their application probably know how to achieve it. Users that don't need it or don't know what it is will only get confused, much like I was at the start of this issue.