jgromes/RadioLib #36: Non-blocking morse implementation.

rnbokade, 2019-08-16:
We are trying to use task scheduler in our application. However, the blocking functions like delay() and tone() cause problems in timing. Can the morse protocol be remade using millis or some other non-blocking function?

jgromes, 2019-08-17:
> Can the morse protocol be remade (...)

Absolutely. Unforunately, I don't have the time work on it right now, but I will happily accept PRs for this.

jgromes, 2019-09-11:
Hey, is this still relevant? If so, I have a bit more time now, so I could have a look at it.

rnbokade, 2019-09-11:
That would be great.

jgromes, 2019-09-11:
OK, what sort of API do you have in mind? Whatever we come up with must be compatible with the current interface.

jgromes, 2019-09-25:
Closed due to inactivity, feel free to reopen later.

closed 2019-09-25 by jgromes

rnbokade, 2020-03-26:
begin (float base, uint8_t speed=20. bool non_blocking=false) should work as a goo API I guess

jgromes, 2020-03-26:
I was talking more about the `print` functions. "Non-blocking" suggests you would want to do other things while Morse data is being sent. Regardless of whether waiting is implemented via `delay` or `millis`, as long as there's a loop that does nothing but wait, it is a blocking function.

rnbokade, 2020-03-26:
Right.... Will think about it more.