jgromes/RadioLib #37: Get by without ChipSelect?

trycoon, 2019-08-25:
Hi, and thanks for a nice library!
I'm have very few pins to spare on my ESP32, I need to run SPI with the fewest pins possible.
I need to find pins for MOSI, MISO, SCK, but the RST I'm connecting to the ESP32 "Reset" pin.
And what about the "SS" ("CS"), since I'm only running ONE device on my SPI bus, do I really need to set this?? Can't I hardwire SS to VCC or GND on the SX1278 module and enter SS=-1 when initializing the library? All feedback is welcome since I'm a total newbee on LoRa.

jgromes, 2019-08-25:
From SX1278 datasheet, page 80:

> A transfer is always started by the NSS pin going low.

SX1278 (and almost all other SPI devices I have used in the past) require CS falling edge to signal the start of communication, so you can't just tie it directly to GND. I would suggest to try and save pins elsewhere - using multiplexers, for example.

Also, as this is unrelated to the library code, I have to close this as invalid. Use Arduino or Espressif forums to ask general questions about Arduino or ESP32.

closed 2019-08-25 by jgromes

trycoon, 2019-08-27:
Thanks, after making further research I believe you are right, SS going low trigger many SPI devices. I will look for a free pin to use.
How the SX1278 works have little with Arduino and Espressif to do, the question was about how the SX1278 will react and how the library will react when setting SS=-1.

jgromes, 2019-08-27:
I beg to differ - the CS pin is only used in the SPI transfer. The only use in the library is when calling `digitalWrite()` Arduino function during SPI transfer, so it really is a generic Arduino/SPI question.