jgromes/RadioLib #42: SX126x sleep() function takes always 5 seconds

beegee-tokyo, 2019-09-11:
### Feature request:
Add parameter for **_`SX126x::sleep(void)`_** to skip waiting for **_BUSY_** line of SX126x chip.
### Why?
In battery powered systems you don't want to wait for the SX126x chip to go to sleep and instead just send the command and then put yourself to sleep.
### Scenario:
I am working on a code for a battery powered sensor node. 
The node wakes up every 30 seconds, reads a sensor, sends the sensor values over LoRa and goes to sleep again.
Right now I am measuring the time that the node is awake and found something strange.
```cpp
void goToSleep(void)
{
  time_t sendToSleep = millis();
  // Send the LoRa module to sleep
  lora.sleep();
  Serial.printf("Set Lora to sleep took %ldms\n", (millis() - sendToSleep));
  Serial.printf("Sleeping after %ldms\n", (millis() - wakeup));
  // Go back to bed
  time_t awakeTime = millis() - wakeup;
  esp_sleep_enable_timer_wakeup((sleepForMillis - awakeTime) * uS_TO_mS_FACTOR);
  esp_deep_sleep_start();
}
``` 
where **_```wakeup = millis()```_** is done as first thing when entering **_```setup()```_**.
The output is always
```log
=====================================
LoRa sensor node test
=====================================
Temp 29.89
Humid 98.50
Finished reading sensor after 1010ms
CAD finished after 24ms
Transmit finished after 70ms
Set Lora to sleep took 5000ms
Sleeping after 6104ms
```
So putting the SX126x to sleep takes nearly 5 times as long as reading the sensors and sending the data.
I tried to track down why it takes so long and found that in **_```sleep()```_** the command is waiting for **_BUSY_** line to go low before returning. 
When I change the **_```sleep()```_** function to
```cpp
int16_t SX126x::sleep() {
  uint8_t data[] = {SX126X_SLEEP_START_COLD | SX126X_SLEEP_RTC_OFF};
  int16_t state = SPIwriteCommand(SX126X_CMD_SET_SLEEP, data, 1, true);

  // wait for SX126x to safely enter sleep mode
  delayMicroseconds(500);

  return(state);
}
```
forcing the SPIwriteCommand to NOT wait for BUSY going low, I can reduce my awake time by 4 seconds to
```log
=====================================
LoRa sensor node test
=====================================
Temp 30.0
Humid 98.50
Finished reading sensor after 1010ms
CAD finished after 24ms
Transmit finished after 70ms
Set Lora to sleep took 0ms
Sleeping after 1104ms
```
### Proposal
Change
```cpp
int16_t SX126x::sleep() {
  uint8_t data[] = {SX126X_SLEEP_START_COLD | SX126X_SLEEP_RTC_OFF};
  int16_t state = SPIwriteCommand(SX126X_CMD_SET_SLEEP, data, 1);

  // wait for SX126x to safely enter sleep mode
  delayMicroseconds(500);

  return(state);
}
```
to
```cpp
int16_t SX126x::sleep(bool waitForBusy) {
  uint8_t data[] = {SX126X_SLEEP_START_COLD | SX126X_SLEEP_RTC_OFF};
  int16_t state = SPIwriteCommand(SX126X_CMD_SET_SLEEP, data, 1, waitForBusy);

  // wait for SX126x to safely enter sleep mode
  delayMicroseconds(500);

  return(state);
}
```
with a declaration as
```cpp
    /*!
      \brief Sets the module to sleep mode.

      \returns \ref status_codes
    */
    int16_t sleep(bool waitForBusy = true);
```
If you agree that this makes sense, I can submit a pull request for it.

beegee-tokyo, 2019-09-11:
Extension after wondering why it is always 5 seconds to return from sleep().
From the datasheet

> 9.3 Sleep Mode
> In this mode, most of the radio internal blocks are powered down or in low power mode and optionally the RC64k clock
> and the timer are running. The chip may enter this mode from STDBY_RC and can leave the SLEEP mode if one of the
> following events occurs:
> • NSS pin goes low in any case
> • RTC timer generates an End-Of-Count (corresponding to Listen mode)
> **When the radio is in Sleep mode, the BUSY pin is held high.**

Which means when you send the sleep command, BUSY will go high and held high until the chip wakes up again. So the ```SPIwriteCommand``` will always go to timeout with the current implementation.
I want to move this from feature request to bug-fix.

jgromes, 2019-09-11:
You're right, obviously it makes no sesnse to wait for BUSY to drop low in sleep mode. That's probably why I added the `waitForBusy` parameter - and then forgot to actually use it...

jgromes, 2019-09-11:
Should be fixed in a04a376, release later today or tomorrow.

closed 2019-09-11 by jgromes

beegee-tokyo, 2019-09-11:
Thanks!

Let's see what else I stumple upon :wink:

jgromes, 2019-09-11:
More bugs, probably - I was in bit of a hurry when writing the SX126x driver ;)

mmrein, 2019-09-11:
@jgromes 
Before you make a release, I would have an enhancement to Sx126x driver - possibility to set a sync word length in bits. I just have to write down the issue and sum up possible solutions. Probably later on today.