jgromes/RadioLib #43: SX126x - add possibility to set SyncWord length in bits

mmrein, 2019-09-11:
Length of SyncWord on SX126x radios is being set as a value in bits, which is solved in library simply by:

```c++
// update packet parameters
_syncWordLength = len * 8;
state = setPacketParamsFSK(_preambleLengthFSK, _crcTypeFSK, _syncWordLength, _addrComp);
```

As I would like to use this feature, I would consider changing:
```c++
int16_t SX126x::setSyncWord(uint8_t* syncWord, uint8_t len)
```
to:
```c++
int16_t SX126x::setSyncWord(uint8_t* syncWord, uint8_t len, bool lenInBits)
```
for example. Then do obvious things in the code (I'll try to write up actual code later on today).

What do you think?

jgromes, 2019-09-11:
What's the use case? All hardware I've seen so far uses sync words that are whole bytes (i.e. sync word length in bits is divisible by 8).

I don't think it's a good idea - the actual sync word value will always be a byte array, so allowing user to set length in bits opens space for ambiguity - for example, if the user sets sync word to {0xFF, 0xFE} and provides 13 bits as the length - which 13 bits should we take?

Furthermore, in the SX126x hardware, sync word is written into 8-bit SPI registers - again, similar ambiguity appears, because the datasheet doesn't specify which bits will be the actual sync word.

mmrein, 2019-09-11:
Simply said, it would be advantageous to use sync word of 10 bits in some modes of Wireless M-Bus.

But I agree the datasheet doesn't make it very clear. I'm just guessing how it works, so the first thing to do is to test it on actual hardware if/how does it work. Or just ask Semtech about it.

The idea was that user should still be able to use standard "setSyncWord(uint8_t* syncWord, uint8_t len)" format in bytes, but if knows what to do, he would be able to use the length in bits too. I have to refresh my poor c++ knowledge, I'm more familiar with plain C.

jgromes, 2019-09-11:
> Or just ask Semtech about it

Yeah, I wish it was that simple :P Protip: don't even bother with their customer support, the only place you can get some information is on their developer forums, and even then, the info from there is dodgy at best. For example, this thread (see the replies about sync word): https://lora-developers.semtech.com/knowledge-base/forum/viewthread/92/

My general approach is to only add features that can be applied IRL. If you have a well defined use case, we can add it.

Pablo2048, 2019-09-11:
In this case there can be method like ```.setSyncBits(uint8_t sunc, uint8_t bits)``` - name of the method explains all and there is no need to modify or overload existing method IMO...

jgromes, 2019-09-11:
@mmrein mentioned 10-bit sync word, this would only get us up to 8. Getting the full range would be possible with `.setSyncBits(uint64_t sync, uint8_t bits)`, though I'm not sure if Arduino core supports 64-bit numbers.

Pablo2048, 2019-09-11:
Ugh - my bad - I did mistake - ```.setSyncBits(uint8_t *sync, uint8_t bits)``` ...

mmrein, 2019-09-11:
@Pablo2048 
that was the other option I was considering.

Pablo2048, 2019-09-11:
Well, actually even better is ```.setSyncBits(const uint8_t *sync, uint8_t bits)``` . I didn't expect that someone will change sync bit word after setting up the radio.

jgromes, 2019-09-11:
I think a `.setSyncBits(const uint8_t *syncBits, uint8_t numBits)` method fits nicely into the current API, questions are the following:

1. Which bits will be valid when `numBits % 8 != 0`? For example, do we take lower bits?
2. Similar to 1 but with respect to the hardware - out of the 8-bit registers, which bits will be considered valid by the radio?

Obviously we can just dump the `syncBits` array into the registers and set the number of sync bits by SPI command, but I don't like the idea of adding this without knowing what will be the actual sync word sent out by the radio.

mmrein, 2019-09-11:
I'll try to debug both on hardware, without knowing this it would be useless. I've tried some quick patch, but I'm still not receiving what I expect.

Side note: is sync word format on SX127x known? I will use the module to test the SX126x against.

jgromes, 2019-09-11:
On SX127x, LoRa sync word is always single byte - AFAIK it can't be turned off.

mmrein, 2019-09-11:
Yes, I know, I mean if the values to be written in registers are straight forward, or if there is some nonsense like the 0x1424 0x3444 as on SX126x.

jgromes, 2019-09-11:
No, the value you write is used as the sync word - though I haven't verified that what it actualy sends out is correct. The confusion with SX126x stems from the fact that LoRa sync word on SX127x is a single byte, whereas on SX126x it's two bytes, and apparently, word 0x12 will match 0x1424.

On a side note, in my experience, LoRa sync word matching on SX127x and SX126x is rather poor and can't be used to reliably distinguish networks.

mmrein, 2019-09-11:
Ok, thanks for the info, I'll try. I should probably set up a RTL-SDR dongle to check what really is in the air.

mmrein, 2019-09-16:
Hello again. After some technical dificulties I did some basic testing. It really seems to work as I was  expecting. 
### Test setup:
**Transmitter (Sx1276, FSK):**
2-byte sync word 0x 0F 55

**Receiver (Sx1261, FSK):**
16bits sync 0x 0F 55 -> Rx OK
16bits sync 0x 0F 54 -> No Rx
14bits sync 0x 0F 54 -> Rx OK
12bits sync 0x 0F 50 -> Rx OK
12bits sync 0x 0F 05 -> No Rx

jgromes, 2019-09-16:
In that case, could you open a PR with the changes you made? Thanks.

mmrein, 2019-09-16:
Sure.

jgromes, 2019-09-17:
Merged #44

closed 2019-09-17 by jgromes