jgromes/RadioLib #48: Issue with the receving example code on SX1262

vasav123, 2019-09-27:
**Describe the bug**
I'm not sure if this is a bug or I'm missing something. However, when I use the SX1262 module and try to setup a receving example, the begin returns an error code of -20. From what I read on the wiki pages, this error is happening because "User tried to execute modem-exclusive method on a wrong modem." But, I know that the SX1262 supports both Lora mode and FSK mode, and therefore that should just work.

**To Reproduce**
Use the Example code from SX1262_Receive.

**Expected behavior**
I should be able to recevie stuff.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Additional info (please complete):**
 - MCU: STM32F103RB
 - Wireless module type SX1262
 - Arduino IDE version 1.8.9
 - Library version: latest

Thanks for your help!

jgromes, 2019-09-28:
This happens when you try to do something with the LoRa modem while the FSK modem is active (or vice versa) - for example, attempting to change FSK bit rate when LoRa modem is active.

Are you using unmodified receive example?

vasav123, 2019-09-28:
Yes, I'm using the examples without making changes.

jgromes, 2019-09-28:
Unable to replicate with STM32L452RE, could you enable debug mode (uncomment [this line](https://github.com/jgromes/RadioLib/blob/eb4fbe9a3d049f31a398e11ac8634bbc367bff6c/src/TypeDef.h#L10)) and post the output?

vasav123, 2019-09-28:
[image]

jgromes, 2019-09-28:
There seems to be no communication from the module. For comparison sake, this is what the output looks like on my side:

```
08:08:35.977 -> [SX1262] Initializing ... 80	0	AA	
08:08:36.078 -> 96	1	A2	
08:08:36.078 -> 8F	0	A2	0	A2	
08:08:36.112 -> 8A	1	A2	
08:08:36.112 -> 93	20	A2	
08:08:36.112 -> 88	3	A2	16	A2	A	A2	0	A2	0	A2	0	A2	0	A2	
08:08:36.180 -> 2	3	A2	FF	A2	
08:08:36.180 -> 8	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	
08:08:36.248 -> 89	7F	A2	
08:08:36.248 -> 11	0	AA	0	1	
```

What's the wiring? Are you sure the module is working?

vasav123, 2019-09-28:
I think so, I tried to use the transmit example with the same pins and that used to work. Also I've tried the example on 2 different boards but neither works.  I'm testing to see if the transmit example works right now but for some reason, I'm getting the same error on that example also now.

vasav123, 2019-09-28:
in terms of the wiring im using the nucleo F103rb board with the semtech 1262 board which is supposed to work with the Arduino headers

jgromes, 2019-09-28:
Since your module always seems to return value 0x00 during SPI transfer, I strongly suspect hardware issue, either in the connection or in the radio itself.

0x00 (as well as 0xFF) is an illegal value for the chip to retrun, I can add check to the library so that the error code correctly reflects the lack of communication (i.e. return ERR_CHIP_NOT_FOUND instead of ERR_WRONG_MODEM), but that's about it.

vasav123, 2019-09-28:
I guess the only way for me to make sure it hardware is to scope it right? Do you have any other ideas?

jgromes, 2019-09-28:
That's probably the only way to be sure. You could also unplug the shield and connect only the SPI, DIO1 + 2 and BUSY pins using some jumpers, to make sure there's  no additional connection that's causing the issues.

jgromes, 2019-09-28:
Closing for now since it doesn't seem like this a bug in the library. Feel free to reopen later.

closed 2019-09-28 by jgromes

Saba-abiri, 2025-02-09:
I had this issue too and I fixed it!
first of all before any command do it :

set NRESET_LORA to Low
wait for about 200 milli seconds
set NRESET_LORA to High
wait for about 500 milli seconds

then send this command

// SetDIO3asTCXOCtrl SPI Transaction
cmnd[0] = 0x97; // 0x97 is SetDIO3asTCXOCtrl
cmnd[1] = 0x00; // DIO3 outputs 1.6 V to supply the TCXO
cmnd[2] = 0x00;
cmnd[3] = 0x00;

let me know if you fix it 👍

(base on my repository you can see more setting : https://github.com/Saba-abiri/LORA1262/blob/main/LORA1262/Libs/Src/SX1262.c)