jgromes/RadioLib #49: SX1262 RX timeout

johnnytolengo, 2019-09-30:
I have a Dorji SX1262 (868) I can't make it work in RX mode. The TX is working "SX126x_Transmit.ino" using "lora.setTCXO(1.6);"

ESP32 with Dorji DRF1262 (DIO2 not connected)
My RX sketch (not working):

`
#include <Arduino.h>
#include <WiFi.h>
#include <RadioLib.h>

time_t updateTime;
// Flag for sending finished
bool sendDone = false;

SPIClass *loraSPI = new SPIClass(HSPI);

//                      NSS DIO1 DIO2 BUSY
SX1262 lora = new Module(15, 22, -1, 21, *loraSPI);
int PIN_LORA_RESET = 19;

#define ssid = "MyWiFi"
#define pass = "qwerty123"
WiFiClient client;
struct tm timeinfo;
uint8_t dataMsg[6];
char ackMsg[] = "ACK";
long timeSinceLastPacket = 0;
uint16_t nodeId = 0;
char timeStamp[32];

void setup()
{
  
  pinMode(PIN_LORA_RESET, OUTPUT);
  digitalWrite(PIN_LORA_RESET, LOW);
  delay(100);
  digitalWrite(PIN_LORA_RESET, HIGH);

  Serial.begin(115200);

// initialize SX1262 
  // carrier frequency:           868.0 MHz
  // bandwidth:                   125.0 kHz
  // spreading factor:            7
  // coding rate:                 5
  // sync word:                   0x1424 (private network)
  // output power:                22 dBm
  // current limit:               60 mA
  // preamble length:             8 symbols
  // CRC:                         enabled
 
 // Initialize LoRa
  Serial.println(F("[SX1262] Initializing ... "));
 
  int state = lora.begin(868.0, 125.0, 7); //SF7
 
  if (state == ERR_NONE) {
    Serial.println(F("LORA  init success!"));
  } else {
    Serial.print(F("init failed, code "));Serial.println(state);
  }

  if (lora.setTCXO(1.6) == ERR_INVALID_TCXO_VOLTAGE)
  {
    Serial.println(F("Selected TCXO voltage is invalid for this module!"));
  }
}

void loop() {
  Serial.print(F("[SX1262] Waiting for incoming transmission ... "));

  String str;
  int state = lora.receive(str);

  if (state == ERR_NONE) {
    // packet was successfully received
    Serial.println(F("success!"));

    // print the data of the packet
    Serial.print(F("[SX1262] Data:\t\t")); Serial.println(str);

    // print the RSSI (Received Signal Strength Indicator)
    // of the last received packet
    Serial.print(F("[SX1262] RSSI:\t\t")); Serial.print(lora.getRSSI()); Serial.println(F(" dBm"));

    // print the SNR (Signal-to-Noise Ratio)
    // of the last received packet
    Serial.print(F("[SX1262] SNR:\t\t")); Serial.print(lora.getSNR()); Serial.println(F(" dB"));

  } else if (state == ERR_RX_TIMEOUT) {
    // timeout occurred while waiting for a packet
    Serial.println(F("timeout!"));

  } else if (state == ERR_CRC_MISMATCH) {
    // packet was received, but is malformed
    Serial.println(F("CRC error!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code ")); Serial.println(state);

  }
}
`

Output of that sketch:
`[SX1262] Waiting for incoming transmission ... timeout!
[SX1262] Waiting for incoming transmission ... timeout!
[SX1262] Waiting for incoming transmission ... timeout!
[SX1262] Waiting for incoming transmission ... timeout!
... 
`

By the way I have another device transmitting over 868Mhz SF7 "helloworld" (tested and working)

Any Idea how to fix it?

jgromes, 2019-09-30:
Timeout simply means no packet was received during the reception window. DIO2 is not required for basic transmit/receive operation. That being said, way too much information is missing here:

This is not a minimal sketch - it contains code that is completely unrelated to the issue (e.g. the WiFi library). What happens when you try to run the receive example, with SPI + TCXO configuration and the correct pins?

> By the way I have another device transmitting over 868Mhz SF7 "helloworld" (tested and working)

How did you test it's working? I.e. do you know it's actually transmitting? How far are the devices, what antennas and what power source do they have?

johnnytolengo, 2019-09-30:
@jgromes that sketch is what I am testing for RX, is true it is dirty with unused variables and other things. 
The SPI runs correctly but DIO1 is not changing at all I mean that it is always in LOW state. 
My sender device is ESP32 +SX1276, I tested it with another "ESP32 +SX1276" in RX mode and works.
I am using the DRF1262T (without DIO2 connection).
My working TX sketch in this HW configuration:

```
 /*
   RadioLib SX126x Transmit Example

   This example transmits packets using SX1262 LoRa radio module.
   Each packet contains up to 256 bytes of data, in the form of:
    - Arduino String
    - null-terminated char array (C-string)
    - arbitrary binary data (byte array)

   Other modules from SX126x family can also be used.

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>

// SX1262 has the following connections:
// NSS pin:   10
// DIO1 pin:  2
// DIO2 pin:  3
// BUSY pin:  9

SPIClass *loraSPI = new SPIClass(HSPI);

SX1262 lora = new Module(15, 22, -1, 21, *loraSPI);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1262 lora = RadioShield.ModuleA;

void setup() {
  Serial.begin(115200);

  // initialize SX1262 with default settings
  Serial.print(F("[SX1262] Initializing ... "));
  // carrier frequency:           434.0 MHz
  // bandwidth:                   125.0 kHz
  // spreading factor:            9
  // coding rate:                 7
  // sync word:                   0x1424 (private network)
  // output power:                14 dBm
  // current limit:               60 mA
  // preamble length:             8 symbols
  // CRC:                         enabled
  int state = lora.begin(868.0, 125.0, 7, 5, 0x1424, 22, 8);
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  

  state = lora.setTCXO(1.6);
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // NOTE: Some SX126x modules use TCXO
  //       (Temprature-Compensated Crystal Oscillator).
  //       To be able to use these modules, TCXO
  //       control must be enabled by calling
  //       setTCXO() and specifying the reference
  //       voltage.
  
  /*
    Serial.print(F("[SX1262] Setting TCXO reference ... "));
    // enable TCXO
    // reference voltage:           1.6 V
    // timeout:                     5000 us
    state = lora.setTCXO(1.6);
    if (state == ERR_NONE) {
      Serial.println(F("success!"));
    } else {
      Serial.print(F("failed, code "));
      Serial.println(state);
      while (true);
    }
  */
  
}

void loop() {
  Serial.print(F("[SX1262] Transmitting packet ... "));

  // you can transmit C-string or Arduino string up to
  // 256 characters long
  // NOTE: transmit() is a blocking method!
  //       See example SX126x_Transmit_Interrupt for details
  //       on non-blocking transmission method.
  int state = lora.transmit("Hello World!");

  // you can also transmit byte array up to 256 bytes long
  /*
    byte byteArr[] = {0x01, 0x23, 0x45, 0x56, 0x78, 0xAB, 0xCD, 0xEF};
    int state = lora.transmit(byteArr, 8);
  */

  if (state == ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F("success!"));

    // print measured data rate
    Serial.print(F("[SX1262] Datarate:\t"));
    Serial.print(lora.getDataRate());
    Serial.println(F(" bps"));

  } else if (state == ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 256 bytes
    Serial.println(F("too long!"));

  } else if (state == ERR_TX_TIMEOUT) {
    // timeout occured while transmitting packet
    Serial.println(F("timeout!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }

  // wait for a second before transmitting again
  delay(1000);
}
```

johnnytolengo, 2019-09-30:
The testing RX sketch:

```
#include <Arduino.h>
#include <WiFi.h>
#include <RadioLib.h>

//https://learn.circuit.rocks/lora-gateway
//https://github.com/jgromes/RadioLib/issues/12

SPIClass *loraSPI = new SPIClass(HSPI);
//                      NSS DIO1 DIO2 BUSY
SX1262 lora = new Module(15, 22, -1, 21, *loraSPI);
int PIN_LORA_RESET = 19;

void setup()
{
  
  pinMode(PIN_LORA_RESET, OUTPUT);
  digitalWrite(PIN_LORA_RESET, LOW);
  delay(100);
  digitalWrite(PIN_LORA_RESET, HIGH);

  Serial.begin(115200);
  Serial.println(F("[SX1262] Initializing ... "));

  int state = lora.begin(868.0, 125.0, 7);

  if (state == ERR_NONE) {
    Serial.println(F("LORA  init success!"));
  } else {
    Serial.print(F("init failed, code "));Serial.println(state);
  }

  if (lora.setTCXO(1.6) == ERR_INVALID_TCXO_VOLTAGE)
  {
    Serial.println(F("Selected TCXO voltage is invalid for this module!"));
  }
}

void loop() {
  Serial.print(F("[SX1262] Waiting for incoming transmission ... "));

  String str;
  int state = lora.receive(str);

  if (state == ERR_NONE) {
    Serial.println(F("success!"));
    Serial.print(F("[SX1262] Data:\t\t")); Serial.println(str);

    Serial.print(F("[SX1262] RSSI:\t\t")); Serial.print(lora.getRSSI()); Serial.println(F(" dBm"));
    Serial.print(F("[SX1262] SNR:\t\t")); Serial.print(lora.getSNR()); Serial.println(F(" dB"));

  } else if (state == ERR_RX_TIMEOUT) {
    Serial.println(F("timeout!"));

  } else if (state == ERR_CRC_MISMATCH) {
    Serial.println(F("CRC error!"));

  } else {
    Serial.print(F("failed, code ")); Serial.println(state);
  }
}
```
The ouput of this sketch:
```
[SX1262] Initializing ... 
LORA  init success!
[SX1262] Waiting for incoming transmission ... timeout!
[SX1262] Waiting for incoming transmission ... timeout!
[SX1262] Waiting for incoming transmission ... timeout!
[SX1262] Waiting for incoming transmission ... timeout!
[SX1262] Waiting for incoming transmission ... timeout!
[SX1262] Waiting for incoming transmission ... timeout!
```

johnnytolengo, 2019-09-30:
I think that I found the problem, seems to be around the DIO2 (always in LOW state).
DIO2 is used to switch RX/TX :
[image]

I tried with to change the DIO2 state with out success with this:

```
lora.setDio2AsRfSwitch(true);

  if (lora.startReceive() != ERR_NONE)
  {
    Serial.print("failed: "); Serial.println(state);
  }
```

Any idea?

jgromes, 2019-10-01:
`startReceive()` is a part of the interrupt-based reception approach. So far, all the code you have posted was using the blocking approach with `receive()`, you can't mix the two.

Simply call `lora.setDio2AsRfSwitch(true);` in the setup function of your previous code and let me know the result.

johnnytolengo, 2019-10-01:
Still the same result with lora.setDio2AsRfSwitch(true);

```
#include <Arduino.h>
#include <WiFi.h>
#include <RadioLib.h>

SPIClass *loraSPI = new SPIClass(HSPI);
//                      NSS DIO1 DIO2 BUSY
SX1262 lora = new Module(15, 22, -1, 21, *loraSPI);
int PIN_LORA_RESET = 19;

void setup()
{
  
  pinMode(PIN_LORA_RESET, OUTPUT);
  digitalWrite(PIN_LORA_RESET, LOW);
  delay(100);
  digitalWrite(PIN_LORA_RESET, HIGH);

  Serial.begin(115200);
  Serial.println(F("[SX1262] Initializing ... "));

  int state = lora.begin(868.0, 125.0, 7);

  if (state == ERR_NONE) {
    Serial.println(F("LORA  init success!"));
  } else {
    Serial.print(F("init failed, code "));Serial.println(state);
  }

  if (lora.setTCXO(1.6) == ERR_INVALID_TCXO_VOLTAGE)
  {
    Serial.println(F("Selected TCXO voltage is invalid for this module!"));
  }

  if(lora.setDio2AsRfSwitch(true) != ERR_NONE)
  {
    Serial.print("failed: "); Serial.println(state);
  }

  //if(lora.startReceive() != ERR_NONE)
  //{
  //  Serial.print("failed: "); Serial.println(state);
  //}

  
}

void loop() {

  Serial.print(F("[SX1262] Waiting for incoming transmission ... "));

  String str;
  int state = lora.receive(str);

  if (state == ERR_NONE) {
    Serial.println(F("success!"));
    Serial.print(F("[SX1262] Data:\t\t")); Serial.println(str);

    Serial.print(F("[SX1262] RSSI:\t\t")); Serial.print(lora.getRSSI()); Serial.println(F(" dBm"));
    Serial.print(F("[SX1262] SNR:\t\t")); Serial.print(lora.getSNR()); Serial.println(F(" dB"));

  } else if (state == ERR_RX_TIMEOUT) {
    Serial.println(F("timeout!"));

  } else if (state == ERR_CRC_MISMATCH) {
    Serial.println(F("CRC error!"));

  } else {
    Serial.print(F("failed, code ")); Serial.println(state);
  }
}
```

output:

```
[SX1262] Initializing ... 
LORA  init success!
[SX1262] Waiting for incoming transmission ... timeout!
[SX1262] Waiting for incoming transmission ... timeout!
[SX1262] Waiting for incoming transmission ... timeout!
[SX1262] Waiting for incoming transmission ... timeout!
```
DIO2 always in LOW state

beegee-tokyo, 2019-10-01:
Do you have anything connected to the Dorji SW or DIO2 pins?

The data sheet is not clear about it, but the pin SW is the power supply for the RF switch and might need to be pulled-up or supplied with 3.3V.

jgromes, 2019-10-01:
I have my doubts about the accuracy of Dorji docs - I have a DRF1269T, which is supposed to have the same antenna switch, but it seems to be working without any changes to DIO2 state.

johnnytolengo, 2019-10-01:
I put 3.3V supply to SW pin and still the same result.
current connection
[image]

The Dorji docs are incomplete as @jgromes  says.

Interesting was that the in TX mode and the SW pin unconnected it worked.

beegee-tokyo, 2019-10-01:
I would try to change 
```
lora.setTCXO(1.6);
```
to a higher value. I am using 2.4V for the eByte E22 modules. 

Other than that, I am glueless, but curious if you can solve it. I was thinking to get some of these modules, because they have FCC and IC, while eByte E22-900M22S has no certification at all.

johnnytolengo, 2019-10-01:
with `lora.setTCXO(2.4)  ` the same result :(
DIO2 never changed his state.

jgromes, 2019-10-01:
> Interesting was that in TX mode and the SW pin unconnected it worked.

Do you know for sure that it works? I.e. can you verify data is actually being sent, for example using SDR or known working receiver?

mmrein, 2019-10-01:
BTW I first started with blocking example and had timeouts too, but I supposed that my transmitter just didn't catch the receive window, so I went straight for the interrupt based example as I wanted to listen infinitely anyways. That worked well (using startReceive(SX126X_RX_TIMEOUT_NONE) or startReceive(SX126X_RX_TIMEOUT_INF) ). 

Few differences on my side, though: SX1261 MBED 868 shield (XTAL), FSK mode, receive only (did not try Tx at all).

johnnytolengo, 2019-10-01:
In my case the thing is the `lora.startReceive(SX126X_RX_TIMEOUT_NONE)` is waiting the DIO1 interrupt and that interrupt never occurrs. I think it must be a hardware setup issue or I don't know...

Here my last attempt:
```
#include <Arduino.h>
#include <WiFi.h>
#include <RadioLib.h>

SPIClass *loraSPI = new SPIClass(HSPI);
//                      NSS DIO1 DIO2 BUSY
SX1262 lora = new Module(15, 22, -1, 21, *loraSPI);
int PIN_LORA_RESET = 19;

volatile bool receivedFlag = false;
volatile bool enableInterrupt = true;

void setFlag(void) {
  if(!enableInterrupt) {
    return;
  }
  receivedFlag = true;
}

void setup()
{
  
  pinMode(PIN_LORA_RESET, OUTPUT);
  digitalWrite(PIN_LORA_RESET, LOW);
  delay(100);
  digitalWrite(PIN_LORA_RESET, HIGH);

  Serial.begin(115200);
  Serial.println(F("[SX1262] Initializing ... "));

  int state = lora.begin(868.0, 125.0, 7);

  if (state == ERR_NONE) {
    Serial.println(F("LORA  init success!"));
  } else {
    Serial.print(F("init failed, code "));Serial.println(state);
  }

  
  if (lora.setTCXO(2.4) == ERR_INVALID_TCXO_VOLTAGE)
  {
    Serial.println(F("Selected TCXO voltage is invalid for this module!"));
  }

  lora.setDio1Action(setFlag);

  if(lora.setDio2AsRfSwitch(true) != ERR_NONE)
  {
    Serial.print("failed: "); Serial.println(state);
  }
 
  if(lora.startReceive(SX126X_RX_TIMEOUT_NONE) != ERR_NONE)
  {
    Serial.print("failed: "); Serial.println(state);
  }
}

void loop() {

  if(receivedFlag) {
    
    enableInterrupt = false;
    receivedFlag = false;

    String str;
    int state = lora.readData(str);

    if (state == ERR_NONE) {
      Serial.println(F("success!"));
      Serial.print(F("[SX1262] Data:\t\t")); Serial.println(str);

      Serial.print(F("[SX1262] RSSI:\t\t")); Serial.print(lora.getRSSI()); Serial.println(F(" dBm"));
      Serial.print(F("[SX1262] SNR:\t\t")); Serial.print(lora.getSNR()); Serial.println(F(" dB"));

    } else if (state == ERR_RX_TIMEOUT) {
      Serial.println(F("timeout!"));

    } else if (state == ERR_CRC_MISMATCH) {
      Serial.println(F("CRC error!"));

    } else {
      Serial.print(F("failed, code ")); Serial.println(state);
    }
  }

  enableInterrupt = true;
}

```

jgromes, 2019-10-01:
Checked with SX1278 transmitter and DRF1268T + DRF1262G receivers, no issues in both blocking and interrupt mode. I don't think it's a problem in the library, so I will have to close this issue for now.

Still, I am curious what's the actual problem here, as I'm using the Dorji modules a lot. Please let me know here if you manage to figure this out - or feel free to reopen if it turns out it's in the library after all.

closed 2019-10-01 by jgromes

johnnytolengo, 2019-10-01:
@jgromes Could you share to me the DRF1268T RX Sketch? to take a look if I am doing something wrong. I will let you know after solve this issue.

jgromes, 2019-10-01:
I used the basic unedited SX126x_Receive and ReceiveInterrupt, the only difference was enabled TCXO with 1.6V reference.

naggie, 2019-12-11:
Sorry to bring up an old issue again, it seems that DIO2 isn't required according to this but is DIO1 to do basic blocking receives and transmissions? As far as I can see the busy line (known as _int0)  will be sufficient for transmission, but it seems _int1 (DIO1) is used for receiving even in blocking mode. If I'm right is this a hardware limitation or implementation limitation?

I'm happy to do a PR to the API reference to explain this when I know. Thanks.

jgromes, 2019-12-12:
@naggie I'm not sure I understand the question - _int0 is DIO1 in all the examples, not BUSY.

In Tx mode, BUSY line will go low when PA ramp-up is finished, not when transmission is done - that is handled by DIO1. In Rx mode, BUSY line goes low when Rx setup is finished and the radio is ready to receive data. DIO1 goes high when full LoRa packet is received.

naggie, 2019-12-12:
Ah -- so the DIO1 line is required to receive and transmit even if not in callback mode. I assume it doesn't have to be an edge interruptible pin on the Arduino, as the code appears to poll in either case.

jgromes, 2019-12-12:
That's right, unless you're using interrupts, DIO doesn't have to be connected to pin with external interrupt on MCUs like ATmega.

In theory you could completely without DIOs, but polling the state of a single pin is a lot quicker and easier than spamming SPI commands ;)

naggie, 2019-12-12:
Brilliant, that makes things clear. Thanks @jgromes

SmartTommy, 2021-09-15:
@johnnytolengo ,have you solved it? i have the same problems, i have three different Modules with the SX1262, two of this works fine, only one never received the signal.

johnnytolengo, 2021-09-15:
@SmartTommy , I don't remember very well but for sure it was a hardware issue.

ricin48, 2025-02-04:
[url=https://hizliresim.com/muwas5a][img]https://i.hizliresim.com/muwas5a.PNG[/img][/url]

[url=https://hizliresim.com/4y44ru4][img]https://i.hizliresim.com/4y44ru4.PNG[/img][/url]

yukarıda paylaştığınız şemada sx1262 nin çıkışındaki PE4259  RF switch entegresinin datasheetini incelediğinizde rx tx mod için hangi kanalı nasıl anahtarladığını anlatıyor..