jgromes/RadioLib #51: SX126x CAD

laughingboyjj, 2019-10-31:
Hi friends
I have been paying attention to your project! When I used the SX1261, I encountered such a problem: when the parameters are set, the CAD is turned on, and the CAD Detect interrupt occurs to enable the receiving mode.

If I use CAD Only mode, it will receive normally, but the packet loss rate is very high;
If you are using the CAD Rx mode, rxtimeout will be generated continuously.
I hope to seek some help, thank you.

jgromes, 2019-10-31:
I'm not sure what you mean - could you poste your code?

laughingboyjj, 2019-11-01:
Please see my GitHub->repositories->LoRa, which is a CAD project with a high packet loss rate. I can't make the SX1261 use CAD function and work well.

jgromes, 2019-11-01:
Your code is not using RadioLib, rather it seems to be using https://github.com/beegee-tokyo/SX126x-Arduino

I didn't write that library, so I can't really help.

closed 2019-11-01 by jgromes