jgromes/RadioLib #52: sx1278 signal strength related functions

retzbeats, 2019-10-31:
Hi jgromes! we are actually using the sx1278 for 433MHz FSK with RadioLib, but we need the signal strength information, we have already check the register map for FSK in the sx1278 and RSSI value and RSSI configuration is available. Do you have a function to read this in the receiver? or can you make that available? The parameters related to RSSI available from the register map are:

- RssiOffset
- RssiSmoothing
- RssiCollisionThreshold
- RssiThreshold
- RssiValue
- TimeoutRxRssi
- Rssi

Regards

retzbeats, 2019-10-31:
ok I understand your concern. You can also add RSSI configuration functions? 
Thanks for your help.

jgromes, 2019-10-31:
> You can also add RSSI configuration functions?

You mean just things in RegRssiConfig (offset and smoothing) - which affect RSSI measurement - or also configuration related to RSSI interrupt? Currently the library has no support for generating interrupt based on RSSI value.

What's the use case? I'm generally against implementing functionality "just because the ship can do it".

jgromes, 2019-10-31:
OK, will try to add this in the next few days.

jgromes, 2019-10-31:
Had a bit of extra time today, I added it to LoRaLib (since that's a standalone SX127x driver from RadioLib) - could you please test it out there?

retzbeats, 2019-10-31:
Yes I will test with that and let you know. Thanks @jgromes .

retzbeats, 2019-11-01:
Hi @jgromes could you please put here the link to the new function code?

jgromes, 2019-11-01:
Ah, sorry, forgot to push the commit yesterday. The function to measure RSSI is still `getRSSI()`, though now it returns float. To set RSSI measurement properties, use [setRSSIConfig](https://jgromes.github.io/LoRaLib/class_s_x127x.html#ad3955f85f456edae772a51025a19029b)

retzbeats, 2019-11-01:
Hi @jgromes we have tested the code and found the following:
- The register you used for RssiValue is the one that is used in the LoRa mapping the one that should be used is the 0x11 for FSK.
- On LoRaLib the receiver code freeze with interruptions and only reads one message.  We have to add the functions you made to RadioLib to make it receive multiple messages working with the interruptions.
- The RSSI that you calculate freezes in one value when we change the RSSI measurement configuration. When we don't use that function the value changes. 

Following is the code we used for the receiver. We'll be waiting for your comments and thanks for your help.
```
/*********
  Modified from the examples of the Arduino LoRa library
  More resources: https://randomnerdtutorials.com
*********/

#include <SPI.h>
/*
   RadioLib SX127x Receive with Interrupts Example

   This example listens for LoRa transmissions and tries to
   receive them. Once a packet is received, an interrupt is
   triggered. To successfully receive data, the following
   settings have to be the same on both transmitter
   and receiver:
    - carrier frequency
    - bandwidth
    - spreading factor
    - coding rate
    - sync word

   Other modules from SX127x/RFM9x family can also be used.

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>

// SX1278 has the following connections:
// NSS pin:   10
// DIO0 pin:  2
// DIO1 pin:  3
//SX1278 lora = new Module(5, 2, 32);
SX1278 sx1278 = new Module(5, 2, 32);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1278 lora = RadioShield.ModuleA;

// flag to indicate that a packet was received
volatile bool receivedFlag = false;

// disable interrupt when it's not needed
volatile bool enableInterrupt = true;

// this function is called when a complete packet
// is received by the module
// IMPORTANT: this function MUST be 'void' type
//            and MUST NOT have any arguments!
void setFlag(void) {
  // check if the interrupt is enabled
  if(!enableInterrupt) {
    return;
  }

  // we got a packet, set the flag
  receivedFlag = true;
}
void setup() {
  Serial.begin(115200);

  // initialize SX1278 with default settings
  Serial.print(F("[SX1278] Initializing ... "));
  // carrier frequency:           434.0 MHz
  // bandwidth:                   125.0 kHz
  // spreading factor:            9
  // coding rate:                 7
  // sync word:                   0x12
  // output power:                17 dBm
  // current limit:               100 mA
  // preamble length:             8 symbols
  // amplifier gain:              0 (automatic gain control)
  //int state = lora.begin(433);
  int state = sx1278.beginFSK(433.91F, 41.667F, 26.37F, 125.0F, 17, 200 ,0.3F,false);
  //int16_t SX1278::beginFSK(float freq, float br, float freqDev, float rxBw, int8_t power, uint8_t currentLimit, uint16_t preambleLength, bool enableOOK) {
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // set the function that will be called
  // when new packet is received
  sx1278.setDio0Action(setFlag);

  // start listening for LoRa packets
  Serial.print(F("[SX1278] Starting to listen ... "));
  state = sx1278.startReceive();
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // if needed, 'listen' mode can be disabled by calling
  // any of the following methods:
  //
  // lora.standby()
  // lora.sleep()
  // lora.transmit();
  // lora.receive();
  // lora.scanChannel();
  sx1278.setRSSIConfig(7,3);
}

void loop() {

  // print RSSI (Received Signal Strength Indicator)
  Serial.print(F("[SX1278] RSSI2:\t\t"));
  Serial.print(sx1278.getRSSI());
  Serial.println(F(" dBm"));
  // check if the flag is set
  if(receivedFlag) {
    // disable the interrupt service routine while
    // processing the data
    enableInterrupt = false;

    // reset flag
    receivedFlag = false;

    // you can read received data as an Arduino String
    String str;
    int state = sx1278.readData(str);

    // you can also read received data as byte array
    /*
      byte byteArr[8];
      int state = lora.receive(byteArr, 8);
    */

    if (state == ERR_NONE) {
      // packet was successfully received
      Serial.println(F("[SX1278] Received packet!"));

      // print data of the packet
      Serial.print(F("[SX1278] Data:\t\t"));
      Serial.println(str);

      // print RSSI (Received Signal Strength Indicator)
      Serial.print(F("[SX1278] RSSI:\t\t"));
      Serial.print(sx1278.getRSSI());
      Serial.println(F(" dBm"));

      // print SNR (Signal-to-Noise Ratio)
      Serial.print(F("[SX1278] SNR:\t\t"));
      Serial.print(sx1278.getSNR());
      Serial.println(F(" dB"));

      // print frequency error
      Serial.print(F("[SX1278] Frequency error:\t"));
      Serial.print(sx1278.getFrequencyError());
      Serial.println(F(" Hz"));

     /* float packet_length;

      packet_length = (float) sx1278.getPacketLength(true);

      printf("packet_length (bytes):%f\n",packet_length);*/

    } else if (state == ERR_CRC_MISMATCH) {
      // packet was received, but is malformed
      Serial.println(F("[SX1278] CRC error!"));

    } else {
      // some other error occurred
      Serial.print(F("[SX1278] Failed, code "));
      Serial.println(state);

    }

    // we're ready to receive more packets,
    // enable interrupt service routine
    enableInterrupt = true;
  }

}

```

jgromes, 2019-11-02:
> The register you used for RssiValue is the one that is used in the LoRa mapping the one that should be used is the 0x11 for FSK.

Fixed, thanks.

> On LoRaLib the receiver code freeze with interruptions and only reads one message. We have to add the functions you made to RadioLib to make it receive multiple messages working with the interruptions.

Which functions do you mean?

> The RSSI that you calculate freezes in one value when we change the RSSI measurement configuration. When we don't use that function the value changes.

Looks like this is unrelated to RSSI configuration, the receiver must be in listen mode to measure RSSI. I added the mode change to the getRSSI function and it seems to be working.

jgromes, 2019-11-17:
Closed due to inactivity - will port from LoRaLib on the next release.

closed 2019-11-17 by jgromes