jgromes/RadioLib #53: How can I change decode model for FSK transmition in SX127X?

moreno-daniel, 2019-11-01:
How can I change decode model for FSK transmition in SX127X? Because, I need to transmit data with Manchester decode

jgromes, 2019-11-01:
What's the use case?

moreno-daniel, 2019-11-01:
The use case is that I am working in a FSK Manchester receptor for pressure sensors

jgromes, 2019-11-01:
Can you elaborate? What sort of receiver/pressure sensor?

LoRaLib/RadioLib (same SX127x driver) doesn't support encoding configuration right now. It should be pretty straight-forward to add it, e.g. something like `int16_t SX127x::setEncoding(uint8_t encoding)`. For encoding value 0 it will set encoding to NRZ, 1 will set it to Manchester and 2 to whitening.

I can add it, but it will be in LoRaLib before it is ported to RadioLib on next LoRaLib update, sou you will have to test using LoRaLib.

moreno-daniel, 2019-11-01:
I am want to recive information of a pressure sensor in a SX127X module. The pressure sensor is for tires, more especific is the "NXP TPMS (Tire Pressure Monitoring Sensors)". At moment, I am design the hardware and paralel I am working at code. The idea is create a protopite low cost for TPMS.

So, for my It would be a great help your colaboration adding this function. Thanks friend.

jgromes, 2019-11-01:
OK, how do you know the sensor is using Manchester encoding?

moreno-daniel, 2019-11-05:
Yes off course, the dasheet link is: http://www.farnell.com/datasheets/2618548.pdf and please and look at page 95 of document

moreno-daniel, 2019-11-05:
I belive that you have a mistake at the new function ```int16_t SX127x::setEncoding(uint8_t encoding)```. The correct form at the function is:

```  
switch(encoding) {
    case 0:
      return(_mod->SPIsetRegValue(SX127X_REG_PACKET_CONFIG_1, SX127X_DC_FREE_NONE, 6, 5));
    case 1:
      return(_mod->SPIsetRegValue(SX127X_REG_PACKET_CONFIG_1, SX127X_DC_FREE_MANCHESTER, 6, 5));
    case 2:
      return(_mod->SPIsetRegValue(SX127X_REG_PACKET_CONFIG_1, SX127X_DC_FREE_WHITENING, 6, 5));
    default:
      return(ERR_INVALID_ENCODING);
  }
```

But, I have changed this and doesn't work. At the moment, I have a RTL-SDR for sniffer the signal and decoding. I use the software URH and this has the next parameters Samples/Symbol and Bits/Symbol. Do you know anything about this?

jgromes, 2019-11-06:
> I belive that you have a mistake

Oops, copy-paste issue. Thanks for pointing that out, fixed now.

> But, I have changed this and doesn't work

What does that mean? Does it not use the selected encoding? Does it not transmit at all? I haven't used URH before, so I'm afraid I can't help you with that.

jgromes, 2019-11-19:
Closed due to inactivity, will be ported to RadioLib on next LoRaLib release.

closed 2019-11-19 by jgromes