jgromes/RadioLib #54: The length byte is optional and should be written/read only if the module is in variable-length mode

phretor, 2019-11-05:
**IMPORTANT: Check the wiki**
Before submitting new issue, please check the [Wiki](https://github.com/jgromes/RadioLib/wiki) and the [API documentation](https://jgromes.github.io/RadioLib/). You might find a solution to your issue there. 

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

jgromes, 2019-11-06:
Variable length mode is the only supported mode for all FSK/LoRa radio modules in RadioLib, because it provides the easiest integration into the high-level API - that's why length byte is always written/read. Therefore, changes in #55 would only make sense if modes other than variable length were available to the user.

I don't plan on adding fixed or unlimited length mode in the near future, but I will accept PRs for that purpose - provided that everything integrates nicely into the high-level API. There was an effort to add support for unlimited packets for SX127x in LoRaLib (https://github.com/jgromes/LoRaLib/pull/66), but ultimately it went nowhere.

> is there any reason why the filter reg value is not cached on the MCU memory?

Most register values aren't cached - it's the old size vs. speed dilema.

phretor, 2019-11-06:
Thanks for considering this.

I wish I had found about RadioLib before, at least before I decided that it was time to for RadioHead to remove the packet-structure assumptions (i.e., header with to/from addresses) they have in there. For this reason I started porting RadioHead code into my [RadioHAL](https://github.com/trendmicro/RadioHAL). I needed a clean, think HAL for [RFQuack](https://github.com/trendmicro/RFQuack). At this point, looking at the radio abstraction alone, the only difference in the approachs of RadioHAL and RadioLib, is the user access to variable/fixed length mode. But of course RadioLib has way many more radios supported. So, I'm considering using RadioLib instead of RadioHAL, but I'm still in the phase where I want to get used to it.

> I don't plan on adding fixed or unlimited length mode in the near future, but I will accept PRs for that purpose

Do the changes in #55 qualify as "acceptable" for your standards?

jgromes, 2019-11-06:
I will have to take a look at how will fixed length packets affect PhysicalLayer.

There's a simple rule when adding new features - if there's a real world use for it, it will likely be added. What's the use case for fixed length packets?

phretor, 2019-11-06:
> What's the use case for fixed length packets?

When developing protocol sniffers, one needs to make zero assumptions on the packet structure, and just receive anything (sometimes even setting the sync words to null and filtering noise in post processing). Some protocols (see DMX512 over RF, just to name the first one that comes to mind) use fixed packet length.

jgromes, 2019-11-06:
Checked PhysicalLayer, there's a minor issue with `getPacketLength` in fixed packet mode that needs to be resolved (see the review comments).

One thing that I don't like about this is that packet mode methods change behavior of transmit() method, which could have an impact once I finally get to implementing protocols that use PhysicalLayer (e.g. LoRaWAN or AX.25) - but I guess the only other option would be to keep separate methods for variable and fixed packets, which seems even worse.

Please take a look at the review comments, if those can be resolved, I will be happy to merge. Also, PRs with packet mode configuration for the other modules (SX127x, SX126x and RF69) would be very much appreciated.

phretor, 2019-11-11:
@jgromes do you want to keep this open until all modules are covered by the change?

jgromes, 2019-11-11:
It's related to all radio modules, so probably best to keep it open until it's added to SX127x and SX126x.

jgromes, 2019-11-22:
Packet mode configuration is implemented for SX127x and SX126x in a8b3823bc309b463a46b02caeb6a85f5e63a6716, closing.

closed 2019-11-22 by jgromes