jgromes/RadioLib #60: Yield-based vs multi-core approach to async TX

phretor, 2019-11-11:
**Is your feature request related to a problem? Please describe.**
No, pure enhancement.

**Describe the solution you'd like**
The `transmit` blocks the core until TX is fully completed. Is this based on the assumption that the WiFi (for instance) and other tasks will run independently on CPU0? Have you considered reading the radio state *before* starting a transmission and don't block but use IRQs to mark a successfull TX?

**Describe alternatives you've considered**
Using ESP8266 and some of Arduino's `yield`.

Example: Example: https://github.com/trendmicro/RadioHAL/blob/master/src/RH_CC1120.cpp#L274

What do you think? My apologies if there is a discussion board/ML/chat to discuss this prior to open an issue: I haven't found any. In that case, please point me there and just close this issue.

jgromes, 2019-11-11:
`transmit` methods are indeed blocking, but they are just a "wrapper" on top of few non-blocking methods. All radio modules follow the basic approach that `transmit` = `startTransmit` + blocking while loop. You can just as well use `startTransmit` to start the transmission and then handle the waiting or interrupts yourself.

RadioLib was originally written for Arduino Uno, hence no multicore support or considerations.

There's no discussion board, I think GitHub issues are the best place to discuss bugs/new features for a library of this size.

phretor, 2019-11-11:
I’m wondering what will happen if the user calls `startTransmit` fast
enough, such that the previous transmission hasn’t finished. Will
everything just crash, or the TX FIFO will be overwritten before the
previous transmission is finished, or nobody except the chip manufacturer
know?

I’ve never tried, so I’m just thinking aloud :-)

jgromes, 2019-11-11:
I'd assume current transmission will be stopped since standby mode is set prior to tx start. and the FIFO will be overwritten - never tried it though and I don't suggest you do ;)

One possible exception from the above could be SX126x becuase of the BUSY pin, but again - never tried this.

closed 2019-11-12 by phretor