jgromes/RadioLib #67: SX126x non-interrupting scanChannel() alternative

BarryPSmith, 2019-11-13:
**Is your feature request related to a problem? Please describe.**
The SX126x `scanChanne`l function will interrupt a receive in progress that might have been started with `startReceive`, causing the packet that is being received to be dropped.

**Describe the solution you'd like**
A method `isChannelBusy()` that, if in receive mode, will query whether a preamble has been recently detected (see additional context below) and return that there is traffic without interrupting the receive.

If the unit is not in receive mode, `isChannelBusy()` will return the result of `scanChannel()`.

**Describe alternatives you've considered**
Can't really think of much.

**Additional context**
I've already got this written: In `startReceive`, an additonal IRQ flag is set up so we get notification of `Preamble Detected`. A custom ISR is attached to DIO1 that tracks the preamble, and calls the consumer defined function when `Rx Done` is signalled.

`isChannelBusy` looks if a preamble has been detected without a Rx Done recently (there's a timeout for in case we get a false preamble detection or never get Rx Done), and returns that the channel is busy if that's the case.

To avoid breaking existing code, an optional parameter can be added to `startReceive` to indicate whether to attach our own ISR to track preamble detection. I would like to add two other new functions too: `setRxDoneAction` and `setTxDoneAction`; these would supersede `setDio1Action` (maybe that could be declared deprecated?).

Let me know if there's interest and I'll spend the time polishing up my code to put it into a PR.

jgromes, 2019-11-13:
I'm not sure about this, similar to #68, handling the timing of `scanChannel` and `startReceive` is up to the user - technically, if the user wants to trigger `scanChannel` during packet reception, he should be allowed to.

Non-blocking CAD would be nice, but I would prefer implementation similar to non-blocking transmit or receive methods - add something like `SX126x::startChannelScan()` to set up CAD, have it trigger interrupt on DIO2 (since DIO1 is used in both transmit and receive), and add a simple method `SX126x::getChannelScanResult()` that will just read the IRQ flags and return LORA_DETECTED or CHANNEL_FREE.

BarryPSmith, 2019-11-13:
This intention here is not to have it non-blocking but rather non-interrupting.

The proposed feature would still allow the user to call `scanChannel` whenever they please; but now they would also have the option of the less intrusive `isChannelBusy` that doesn't stop the modem from receiving if they have already called `startReceive`.

Right now there is no way the user can know if the modem has detected a preamble and is busy recording a packet. They only get notified on `Rx Done`.

Estimating whether a packet is currently being received requires the following components that are not exposed to the library consumer:
 - Changing the interrupts enabled when calling `startReceive` so that `Preamble Detected` triggers an interrupt.
 - If using the same DIO for multiple IRQs, call `getIrqStatus` and `clearIrqStatus`.

Because the devices don't offer an interface to indicate if they are busy recording a packet, the determination must be an estimate based on how long most packets are take and the last time a preamble was detected.

The motivation is a something like this:
 - User wants to receive data from a distant modem, they also need to periodically send data.
 - User calls `startReceive`
 - Distant modem begins to send user a packet. User's modem begins to receive the packet
 - User now wants to send a packet. If they call `scanChannel`, they will drop their half received packet. If they just transmit, they will drop their half received packet and likely both their packet and the half transmitted packet will not be picked up by anyone.

It's true that this introduces an algorithm into the library that is not really backed up by the hardware, it merely takes records timings from the hardware to guess at what is actually going on.

jgromes, 2019-11-14:
> This intention here is not to have it non-blocking but rather non-interrupting.

Misread the title, sorry.

I think the BUSY line should be able to handle this alone, in section 8.3.1, the datasheet says "In additon to this, the BUSY will also go high to handle its internal IRQ. In this scenario, it is essential to wait for the BUSY line to go low before sending an SPI command (either a “read” or “write” command)." It should be probably verified with an oscilloscope - I learned not trust Semtech's docs ;)

Is the packet dropping an issue you have faced before? I'm trying to figure out how often can this happeon in real-world scenario.

BarryPSmith, 2019-11-14:
The busy line indicates whether the module can receive a command over the SPI interface. It doesn't seem to be affected by whether or not the unit is currently sending or receiving data. I know because in implementing this I have a function that queries and clears the IRQ registers using SPI while the packet is being received. If the busy line was being held high during packet receive, the IRQ requests would time out in the `waitForBusy` loop.

I haven't encountered the issue of dropped packets with the SX1262 yet. But I'm upgrading from a system that was using a TNC modem + handheld receiver (which was working at a similar baud), and I saw collisions pretty regularly there. Given that the condition for a collision on that system is the same as the condition for a dropped packet on this one (One unit wants to Tx halfway through another's message), I'm pretty sure I'll see dropped packets.

Actually, on re-reading the datasheet I noticed there's an option for CAD where it goes into receive if it detects activity. The datasheet doesn't explicitly state what the behaviour is if everything is done halfway through a packet; I'll investigate that if it provides a hardware solution.

jgromes, 2019-11-29:
Closed due to inactivity - let me know if the hardware solution prvoes to be enough.

closed 2019-11-29 by jgromes