jgromes/RadioLib #68: Carrier-sense multiple access (CSMA / being polite)

BarryPSmith, 2019-11-13:
**Is your feature request related to a problem? Please describe.**
The radios don't wait until the channel is clear before transmitting. If a unit starts to transmit halfway through a packet directed to it, it will miss that packet. It might also cause other stations to miss the packet it interrupts, or its packet might not be heard (or both).

**Describe the solution you'd like**
An implementation of p-persistence CSMA.

Configuration parameters might include:
 - bool performCSMA: Whether this feature should be active or whether the unit will transmit regardless.
 - float timeslotPeriod: How large a timeslot to use to perform the p-persistence algorithm. If the consumer does not provide this, RadioLib could use an appropriate value based on the estimated time on air of the packet to send (and the assumption that other packets on the network will be comparable size).
 - float p: probability of transmitting in any given timeslot. if p=1 this reduces to 1-persitence (transmit as soon as the channel is clear).

**Describe alternatives you've considered**
The `isChannelBusy()` proposed in #67 could be the standard function that physical layers expose if they are able to be used with feature. This can be implemented without the complex functionality described there, it just won't be as good.

**Additional context**
https://en.wikipedia.org/wiki/Carrier-sense_multiple_access

I'm going to write this code anyway, if there's interest in including it in this library we can figure out a good interface before I get started.

jgromes, 2019-11-13:
I don't think it's a good idea to add this. At the core, RadioLib is a collection of drivers for radio modules - the library does all the heavy lifting of register configuration etc., and leaves the top level functionality fully exposed to the user - the user can then implement whatever he wants on top of that. This restriction means that only features of the hardware are implemented.

What you're describing seems to be more like a part of some protocol - it's a higher level of abstraction and it's not a feature of any radio module.

BarryPSmith, 2019-11-13:
This is true. It's OSI layer 2 (one layer above physical layer). I thought it might fit in with the other protocols in the library.

BarryPSmith, 2019-11-13:
Yes, I would certainly think to do something along those lines. But the protocol has to know whether there is traffic on the channel in order to work. Currently there is no method on Physical layer like `isChannelBusy`, although all of the hardware units provide something that could be used:
RF9x, SX126x and 127x provide `scanChannel`.
RF69 would have to implement something with the RSSI DIO or register.
nRF24 looks like it needs to expose RPD measurements (datasheet section 6.4).
C1101 can expose Carrier Sense (Datasheet 17.4)

As I mentioned, I don't have much desire to write the code for modules I don't own or use, but I can write the protocol to work on SX126x and that can be expanded in future.

BarryPSmith, 2019-11-14:
Hey nevermind, I wrote the code and it's only 17 lines long and you're right it doesn't need to be a part of RadioLib.

closed 2019-11-14 by BarryPSmith

jgromes, 2019-11-14:
I remain sceptical, sorry. My main issue is that different protocols might handle collisions differently. If at some point we come to the conclusion it would be advantageous to provide information about channel state to the upper layers, your outline for what to use in different modules would be 100% correct.

Also (mainly as a note for your code), `scanChannel` on RF9x and SX127x will only detect LoRa preamble, not the whole packet. In addition, RF9x, SX127x and SX126x have FSK modem, which can't use `scanChannel` at all and must rely on RSSI measurement - not sure if this is relevant to your code, just putting it out there.

Cloolalang, 2020-06-19:
Mysensors library is using it in RFM95_sendWithRetry