jgromes/RadioLib #69: SX126x cannot wake from sleep

BarryPSmith, 2019-11-14:
**Describe the bug**
SX126x modules cannot be woken with the library, instead one must manually wake the unit with a digital write to its select pin.

It looks like in `SX126x::SPItransfer`, moving the line `digitalWrite(_mod->getCs(), LOW);` to run before the `Wait for BUSY` loop will fix this (first sentence in datasheet sentence 8.2.2) but I'm not sure if there's a reason it needs to be where it is.

**To Reproduce**
```
#include <RadioLib.h>

#define LORA_CHECK(A) { auto res = A; if(res) { Serial.print("FAILED " #A ": "); Serial.println(res); } } while(0)

#define BUSY 4
#define DIO1 2
#define DIO2 -1
#define SELECT 9
SX1262 lora = new Module(SELECT, DIO1, DIO2, BUSY);

void setup() {
  Serial.begin(9600);
  delay(10);
  LORA_CHECK(lora.begin());
  LORA_CHECK(lora.sleep());
  //digitalWrite(SELECT, LOW); //Uncomment this and everything works
  LORA_CHECK(lora.standby()); //Fails with -705
  Serial.println("Setup done. If no errors show, everything worked.");
}

void loop() {}
```

**Expected behavior**
The module can be woken from sleep by asking it to return to standby

**Additional info (please complete):**
 - MCU: Ardiuno Nano
 - Wireless module type: SX1262
 - Arduino IDE version:   1.8.10
 - Library version: tip of trunk as of writing. Checkout 300e7f48fddad455c96df

jgromes, 2019-11-14:
I was able to reproduce this, thanks.

You're right, NSS must go low before checking BUSY pin. BUSY should be held high in sleep mode, and since we wait for BUSY to go low at the start of `SX126x::SPItranscation()`, it will never leave the waiting loop, hence the timeout.

I'm not entirely sure why is NSS pulled low after checking BUSY pin - I think it might have been an effort to not start any SPI transactions as long as BUSY is high.

jgromes, 2019-11-14:
Fixed in 3db5134, thanks for reporting!

closed 2019-11-14 by jgromes