jgromes/RadioLib #72: CC1101 goes to Standby after receiving a packet.

Guglio95, 2019-11-19:
Hi Jan,
I've a question about the implementation of the CC1101 receive flow,
Why does the [readData() method ](https://github.com/jgromes/RadioLib/blob/master/src/modules/CC1101.cpp#L286) put the radio back in idle mode after a packet is read from RX FIFO?

I think this way the [example](https://github.com/jgromes/RadioLib/blob/master/examples/CC1101/CC1101_Receive_Interrupt/CC1101_Receive_Interrupt.ino) will not work properly since the radio is put in receive mode only [once](https://github.com/jgromes/RadioLib/blob/master/examples/CC1101/CC1101_Receive_Interrupt/CC1101_Receive_Interrupt.ino#L57) , then data is read using readData whose side effect is to put radio back in IDLE. This way (I suppose) no other packet will be received.

Unfortunately I cannot test it with a real radio at the moment.
Thankyou

jgromes, 2019-11-19:
Well spotted, thanks! `startReceive()` must be called after `readData()` to restart the listen mode. The example needs the same amendment as in #50.

It also looks like this is missing from SX127x and RF69 examples, will fix that too.

closed 2019-11-19 by Guglio95

jgromes, 2019-11-19:
Should be fixed in update 1.8.0