jgromes/RadioLib #76: lilygo ttgo t-deer gives error -12 on "in range" frequency

zanco, 2019-11-26:
Bought 3 units with the same radio label:TTGO 433 470 MHz Modle LoRa32 radio module, SX1278

I have one unit which gives error -12 when I set the LoRa frequency to anything between 432 and 470 MHz and no error when I set 868 - 880 MHz however it does not transmit on that 868 MHz frequency.

The code sends RTTY and LoRa alternating (for radiotracker under a balloon) and RTTY does transmit on 432 MHz without problems. 

`// Default RTTY settings. Change when needed
#define RTTY_FREQUENCY  432.762
#define RTTY_SHIFT 600
#define RTTY_BAUD 75
#define RTTY_STOPBITS 1

// Default LORA settings. Change when needed
#define LORA_FREQUENCY  869.662
#define LORA_BANDWIDTH 125.0
#define LORA_SPREADFACTOR 9
#define LORA_CODERATE 7`

I have no clue why this does happen.
The first hardware board from the same seller does not give this error and does transmit on 432.762 if I set that frequency.

Running the same board with Stuart Robinson tracker software does do tranmist both LoRa and RTTY on 432.762 if I set that frequency.

Any idea what could be the issue ?

jgromes, 2019-11-26:
SX1278 should be limited to 137 to 525 range.

Could you please provide a minimum sketch required to reproduce this?

zanco, 2019-11-27:
I think I can provide "the sketch" which gives the problem, however, the problem only shows on one module, not on the other. I removed the lid and the radiochip shows SX1278. I am now going to test the same sketch on the third module.

jgromes, 2019-11-27:
RadioLib only checks the chip type on startup - frequency range is hard-coded based on the radio type, so I find it rather odd that the same sketch on the same radio chips has different results.

I suggest trying a minimal sketch on both devices, e.g. the following:

```c++
#include <RadioLib.h>

SX1278 lora = new Module(10, 2, 3);

void setup() {
  Serial.begin(9600);
  Serial.println(lora.begin()); // should always return 0
  Serial.println(lora.setFrequency(432.762)); // in range, should return 0
  Serial.println(lora.setFrequency(869.662)); // out of range, should return -12
}

void loop() {
  
}
```

zanco, 2019-11-27:
Thanks,
returns 0, 0, -12 at both modules so works as expected. I did run the settings example with the same result, 434 in range and 869 out of range. So back to "our source" to find out why this happens only at our software.

closed 2019-11-27 by jgromes