jgromes/RadioLib #79: Where can I find a changelog?

BarryPSmith, 2019-12-01:
**Is your feature request related to a problem? Please describe.**
My previously working modules and code didn't work with the latest version of the library, and it was quite difficult to figure out why.

Is there a changelog I can check? I'm very bad at looking for things. Perhaps I didn't see it in the wiki or github pages, but I can't find a changelog.

**Describe the solution you'd like**
When there are breaking changes it would be nice if they were singled out and mentioned somewhere

**Describe alternatives you've considered**
Github provides a list of all check-ins, but it's quite burdensome on users to read all checkins to try to find major changes

**Additional context**
Recent changes mean that client code using SX126x modules must use the three variable module constructor instead of the 4 variable constructor.

Also, the SX126x now turns on TCXO by default. If running with a module that doesn't use TCXO, this means that things suddenly stop working with no discernable indication why (I had to go block out code with `#if 0` / `#endif` to determine where the issue was)

jgromes, 2019-12-01:
I usually write down the major changes for every release in https://github.com/jgromes/RadioLib/releases

However, the commit history is the primary source for the overall development of the repo.

closed 2019-12-01 by jgromes

BarryPSmith, 2019-12-01:
Brilliant! Thank you. How do I get there if someone doesn't provide a direct link? I can't see anything on the main page of the repository. Perhaps if others are as blind as me, a link from the github pages or wiki would increase visibility...

Actually there's a bunch of stuff on github which as far as I can tell the only way to get there is to know about it through other channels.

jgromes, 2019-12-01:
It's probably not the worst idea to link to it from readme.