jgromes/RadioLib #80: SX126x do not check CRC

BarryPSmith, 2019-12-01:
**Describe the bug**
The CRC of incomming packets is not checked. Garbled packets are passed through readData(...) with no indication that the data is bad.

I don't know whether or not Tx packets have a CRC at all, because I only have these units to test with. I am certain that no CRC check is done because at range I was getting corrupted packets with no software errors to indicate they weren't correct (until they failed to process downstream).

I hope to have a PR for this in a short while.

**To Reproduce**
Two sketches, one running receive and one running transmit. The receive sketch is supposed to check CRCs with the call to `lora.setCrc(1)`. The transmit sketch alternatively turns CRCs on and off. I would expect the packets without CRCs to fail. However, all packets are received fine.

(Note: These use 425MHz because I was having problems getting my units to Tx on 433. Change as appropriate for regulatory conditions)
Transmit sketch:
```
#include <RadioLib.h>

#define LORA_CHECK(a) do {\
      auto res = a; \
      if (res != ERR_NONE) \
      { \
        Serial.print("Error performing " #a ": "); \
        Serial.println(res); \
        while(1) {} \
      } else { \
        Serial.println(#a " successful."); \
      }} while(0)

#define DIO1 2
#define BUSY 4
#define SELECT 9

#define FREQ 425.0
#define SF 7
#define BW 62.5
#define CR 5

Module mod(SELECT, DIO1, BUSY);
SX1262 lora = &mod;

void setup() {
  Serial.begin(38400);
  delay(10);
  Serial.println("starting...");
  if (lora.begin(FREQ, BW, SF, CR))
    LORA_CHECK(lora.begin(FREQ, BW, SF, CR));
  LORA_CHECK(lora.setOutputPower(10));
  LORA_CHECK(lora.setCurrentLimit(140));
  LORA_CHECK(lora.setDio2AsRfSwitch());
  LORA_CHECK(lora.transmit("<CALLSIGN>"));
}

void loop() {
  static bool useCrc = true;
  static char message[] = "CrcCheck. State: #";
  
  useCrc = !useCrc;
  LORA_CHECK(lora.standby());
  LORA_CHECK(lora.setCRC(useCrc));
  LORA_CHECK(lora.standby());
  message[17] = '0' + useCrc;
  LORA_CHECK(lora.transmit(message));
  Serial.print("Finished transmitting: \"");
  Serial.print(message);
  Serial.println("\"");
  delay(1000);
}
```

Receive sketch:
```
#include <RadioLib.h>

#define LORA_CHECK(a) do {\
      auto res = a; \
      if (res != ERR_NONE) \
      { \
        Serial.print("Error performing " #a ": "); \
        Serial.println(res); \
      } } while(0)

#define DIO1 2
#define BUSY 4
#define SELECT 9

#define FREQ 425.0
#define SF 7
#define BW 62.5
#define CR 5

Module mod(SELECT, DIO1, BUSY);
SX1262 lora = &mod;

volatile bool flag;
void setFlag()
{
  flag = true;
}

void setup() {
  Serial.begin(38400);
  if (lora.begin(FREQ, BW, SF, CR))
    LORA_CHECK(lora.begin(FREQ, BW, SF, CR));
    
  LORA_CHECK(lora.setOutputPower(10));
  LORA_CHECK(lora.setCurrentLimit(140));
  LORA_CHECK(lora.setDio2AsRfSwitch());
  LORA_CHECK(lora.setCRC(1));
  lora.setDio1Action(setFlag);
  LORA_CHECK(lora.startReceive(SX126X_RX_TIMEOUT_INF));
}

void loop() {
  char buffer[255];
  if (flag)
  {
    auto len = lora.getPacketLength();
    LORA_CHECK(lora.readData(buffer, 255));
    buffer[len] = 0;
    Serial.print("Received: ");
    Serial.println(buffer);
    LORA_CHECK(lora.startReceive(SX126X_RX_TIMEOUT_INF));
    flag = false;
  }
}
```

**Expected behavior**
A packet with a failing CRC should give an error on `readData`

**Additional info (please complete):**
 - MCU: Arduino Nano
 - Wireless module type SX1262
 - Arduino IDE version  1.8.10
 - Library version 2.0

BarryPSmith, 2019-12-01:
This is a one line fix in #81.

CRC errors are difficult to reliably reproduce. Some library changes are necessary to make this happen: Enable godmode to allow the receiver to set the packet length in advance; and change to implicit header mode. Testing patch:
```
diff --git a/src/TypeDef.h b/src/TypeDef.h
index c9898a9..eacec77 100644
--- a/src/TypeDef.h
+++ b/src/TypeDef.h
@@ -26,6 +26,7 @@
 
 //#define RADIOLIB_DEBUG
 //#define RADIOLIB_VERBOSE
+#define RADIOLIB_GODMODE
 
 // set which Serial port should be used for debug output
 #define RADIOLIB_DEBUG_PORT   Serial
diff --git a/src/modules/SX126x/SX126x.h b/src/modules/SX126x/SX126x.h
index ff25687..64b90be 100644
--- a/src/modules/SX126x/SX126x.h
+++ b/src/modules/SX126x/SX126x.h
@@ -769,7 +769,7 @@ class SX126x: public PhysicalLayer {
     int16_t setTxParams(uint8_t power, uint8_t rampTime = SX126X_PA_RAMP_200U);
     int16_t setModulationParams(uint8_t sf, uint8_t bw, uint8_t cr, uint8_t ldro = 0xFF);
     int16_t setModulationParamsFSK(uint32_t br, uint8_t pulseShape, uint8_t rxBw, uint32_t freqDev);
-    int16_t setPacketParams(uint16_t preambleLength, uint8_t crcType, uint8_t payloadLength = 0xFF, uint8_t headerType = SX126X_LORA_HEADER_EXPLICIT, uint8_t invertIQ = SX126X_LORA_IQ_STANDARD);
+    int16_t setPacketParams(uint16_t preambleLength, uint8_t crcType, uint8_t payloadLength = 0xFF, uint8_t headerType = SX126X_LORA_HEADER_IMPLICIT, uint8_t invertIQ = SX126X_LORA_IQ_STANDARD);
     int16_t setPacketParamsFSK(uint16_t preambleLength, uint8_t crcType, uint8_t syncWordLength, uint8_t addrComp, uint8_t whitening, uint8_t packetType = SX126X_GFSK_PACKET_VARIABLE, uint8_t payloadLength = 0xFF, uint8_t preambleDetectorLength = SX126X_GFSK_PREAMBLE_DETECT_16);
     int16_t setBufferBaseAddress(uint8_t txBaseAddress = 0x00, uint8_t rxBaseAddress = 0x00);
     uint8_t getStatus();

jgromes, 2019-12-01:
Thanks, out of curiosity - how reliably can you create CRC errors this way? Haven't checked wth hardware yet, but this looks useful to have.

jgromes, 2019-12-01:
Merged #81

closed 2019-12-01 by jgromes

BarryPSmith, 2019-12-01:
By telling the transmitter not to include a CRC, you get an error every time. But only if you set it to implicit header. The hard part was just me not being meticulous enough - it took a while to find the part of the datasheet where it said the implicit header was necessary (actually I found it in the SX1278 datasheet), then a while longer to realise that I had to manually set the packet length.

naggie, 2019-12-11:
Just came across this issue, as I've been looking to enable implicit header mode to reduce latency. Thanks for the hack @BarryPSmith -- I can use that in the mean time. @jgromes  -- perhaps implicit header mode should be an option of the sx126x api. I'll create an issue if you agree -- I could do a PR but I'd like to make sure you're happy with however it's implemented.

Thanks for the shockingly well structured and documented library.

jgromes, 2019-12-11:
@naggie Thanks! Feel free to open a new issue to discuss the implicit mode there.