jgromes/RadioLib #82: Incorrect preamble sync?

observator-x, 2019-12-03:
As far as i noticed using the FSK TX example of radiolib i should receive a preable + sync of:

0x55,
0x2D, 0x01,

0123456789abcdefab5a (data+2byte crc)

but when i capture the packet i get:

0x2a, 0xaa, 0xaa, 0x96, 0x80, 0x84

0123456789abcdefab5a (data+2byte crc)

what could be wrong? i use:

 - Modules [RFM95, ESP8266]
 - Arduino IDE version  [1.8.10]
-  Radiolib latest version

jgromes, 2019-12-03:
You're right that 0x55 should be the default preamble value, just as the default sync word should be 0x2D01. What are you using to capture the packets?

observator-x, 2019-12-03:
LIME-SDR and GNUradio

jgromes, 2019-12-03:
That doesn't tell me a whole lot. Could you please share the exact setup? (blocks used, configuration etc.)?

Also, were you able to verify that the data is correct, for example with a known working transmitter?

observator-x, 2019-12-03:
I receive the signal succesfully with another RFM95 with same setup, also when i record the IQ signal with the SDR and re-broadcast it it sucessfully received.  i can show you my current setup with UHR wich i have currently setup to receive the packets

jgromes, 2019-12-03:
I checked the SPI registers relevant to this (SX127X_REG_SYNC_CONFIG and SX127X_REG_SYNC_VALUE_1 - SX127X_REG_SYNC_VALUE_8) and the values are the expected ones (preamble 0x55, sync word enabled, two bytes, 0x2D 0x01).

I think I know what's going on - you start decoding one bit too late, hence all your data is shifted one bit. That's how 0x55 preamble becomes 0xAA and how 0x2D01 turns into 0x9680. Assuming you're using the default configuration, there's also a length byte between sync word and data bytes. You sent 8 bytes of data, hence the length byte will be 0x08.

The shift is clearly visible here:
```
0x552D0108 = 01010101001011010000000100001000
0xAA968084 = 10101010100101101000000010000100
```

Also, I have RTL-SDR dongle and haven't used UHR before. Could you please share your configuration or point me towards an article/guide to decode FSK data using that software? It would be incredibly helpful in the future.

closed 2019-12-03 by jgromes