jgromes/RadioLib #85: Unstruct Arbitrary binary data

marlko, 2019-12-09:
How can i unstruct the text send to another device?
I send from a nodemcu RadioLib lib to a raspberry pi, but the only thing that i receive is a byte array, and isnt always the same.

Ex: 
[72, 101, 108, 111, 96, 126, 3, 16, 225, 14, 82]
[72, 101, 108, 111, 96, 126, 3, 16, 225, 94, 18]
[72, 101, 108, 111, 96, 126, 3, 16, 225, 94, 30]

jgromes, 2019-12-10:
Which module are you using?

marlko, 2019-12-10:
SX1278 Ra-01 V3

jgromes, 2019-12-10:
No sure I understand the question - the data is sent the exact same way, the only difference is that string-based `receive()` assumes the data is null-terminated C-string. The binary version of that method can be used to send anything.

jgromes, 2019-12-24:
Closed due to inactivity

closed 2019-12-24 by jgromes