jgromes/RadioLib #86: Implicit mode support for Semtech LoRa  SX126x modules

naggie, 2019-12-11:
**Is your feature request related to a problem? Please describe.**
It's not currently possible to put the modem in Implicit header mode which is useful for low latency applications where the payload size is always known.

**Describe the solution you'd like**
I think a `SX126x::setHeaderType)` would work, though I'm not familiar with the library to know this is the best interface.

This would supply `SX126X_LORA_HEADER_EXPLICIT` or `SX126X_LORA_HEADER_IMPLICIT` to `setPacketParams()` where appropriate.

jgromes, 2019-12-12:
As far as I can tell, this shouldn't break compatibility with existing code, since all the parameters required for implicit header (payload length, coding rate and CRC presence) should by configured by the default values of `begin()`.

I think the interface should be similar to the way fixed/variable packet length mode is handled, i.e. use two public methods:

```
int16_t SX126x::implicitHeader();
int16_t SX126x::explicitHeader();
```

both of which call common protected `int16_t SX126x::setHeaderType()`.

Also, SX127x has implicit header mode too - perhaps the same interface should be added to that one as well.

naggie, 2019-12-13:
OK -- I assume calling one of those methods should set a variable which all calls to `setPacketParams` would be called with (whenever called by something else) rather than actually programming the radio at time of call.

Alternatively setPacketParams could use the member variable instead of taking a parameter (which is currently not used). What would you prefer?

jgromes, 2019-12-14:
The issue with this approach is that if the user only wants to change current header mode, he would then have to call some method which calls `setPacketParams` - there is at least one corner case when user could call `explicitHeader` (after having previously used implicit mode), then call some reception method (e.g. `receive`). There would be no call to `setPacketParams`, hence the header mode would remain implicit, even though user thinks explicit mode is being used.

Since there's no documented way to read the current configuration of SX126x, I think the safest approach is to call `setPacketParams`, despite the increased SPI traffic. `_preambleLength` and `_crcType` should be known at the time of call to `implicitHeader` or `explicitHeader`.

Also, length parameter must be added to `startReceive` and `startReceive` common, to set packet length in implicit header mode.

jgromes, 2019-12-28:
Closed due to inactivity - @naggie feel free to reopen later or submit a PR.

naggie, 2019-12-28:
Ok -- I will submit a PR when I return to work.

naggie, 2020-01-09:
> The issue with this approach is that if the user only wants to change current header mode, he would then have to call some method which calls `setPacketParams` - there is at least one corner case [...]

Good point.

> Since there's no documented way to read the current configuration of SX126x, I think the safest approach is to call `setPacketParams`, despite the increased SPI traffic. `_preambleLength` and `_crcType` should be known at the time of call to `implicitHeader` or `explicitHeader`.

So, 
1. Calling `SetHeaderMode` will call `setPacketParams` _and_ set a member variable recording header type
2. The header type should be passed to `setPacketParams` on every other invocation from the member variable

> Also, length parameter must be added to `startReceive` and `startReceive` common, to set packet length in implicit header mode.
OK. This will be a breaking change unless I set an explicit-mode only default. I think a breaking change is best as it results in simpler code.

jgromes, 2020-01-09:
> receive requires a length -- perhaps this should be the precedent.

That's what I meant in my previous comment - fundamentally, implicit header requires length of the packet to be known in advance. I can see two options how to solve this:

1. Adding expected packet length as a parameter to `implicitHeader()` method.
2. Adding expected packet length as a parameter to receive methods.

Option 1 is probably easier to implement, doesn't break compatibility and (IMO) makes more sense from user POV.

naggie, 2020-01-09:
OK -- does option 1 mean that `startReceive` et al don't have to call setPacketParams? And/Or that `_expectedLen` needs to be created as another member variable to cache the value?

jgromes, 2020-01-09:
I think setPacketParams isn't called from any receive method atm, we can leave it like that.

setPacketParams is however called from startTransmit, and it changes payload length. This could be an issue if user does the following:

1. set implicit mode with expected length
2. send a packet with different length than in 1
3. attempt to receive - SX126x will now expect packets with length set in step 2 as opposed to step 1

naggie, 2020-01-09:
Good point. So perhaps the solution is to require the same length and error otherwise? Or just provide a warning in the documentation?

jgromes, 2020-01-09:
It's not really an error, more like an unexpected behavior. The safest solution I can think of is:

1. add expected length parameter to implicit mode method
2. cache the expected length
3. in transmit, let user provide any length
4. in receive & only for implicit mode, call setPacketParams to set expected length

Bit convoluted, but should be safe for every possible combination of transmit/receive/header changes.

naggie, 2020-01-09:
OK, will do. That also solves another problem -- `setPreambleLength` and `setPreambleLength` also call `setPacketParams`.

jgromes, 2020-01-09:
As does `setCRC()` ;)

Side note - this is the reason I really don't like the SX126x's SPI interface, it's needlesly complicated & has almost no advantages over direct register access like SX127x has.

naggie, 2020-01-09:
> As does `setCRC()` ;)

Haha that was what I was trying to paste :)

> Side note - this is the reason I really don't like the SX126x's SPI interface, it's needlesly complicated & has almost no advantages over direct register access like SX127x has.

Interesting -- though sx1276 is a better performing/newer part isn't it? Direct reg access does seem to be a much saner idea.

----

Maybe this new implementation also means that setHeaderType doesn't need to call setPacketParams (assuming the modem defaults to explicit mode)

jgromes, 2020-01-09:
> Maybe this new implementation also means that setHeaderType doesn't need to call setPacketParams (assuming the modem defaults to explicit mode)

That's actually correct - looks like we made it right back to the start :D Though it might be safer to keep the call to setPacketParams, if only to keep with the "expected behavior".

> sx1276 is a better performing/newer part isn't it?

Actually no, SX127x is the older LoRa chip, SX126x is the newer one. Also, not to be confused with SX128x, which is a LoRa radio too, but works in the 2.4GHz band ;) For some reason, Semtech decided to completely rework the SPI interface in their new radio chips.

> Direct reg access does seem to be a much saner idea

100% agree, and here's the best part - SX126x still has direct reg access, only none of the registers are actually documented. I digress - I could go on a rant about all the reasons why SX126x is inferior to SX127x from interface/application standpoint, so better to stop now.

naggie, 2020-01-10:
Ok I've implemented the mechanism, though there's another API problem -- the (blocking) receive method wants a len which appears to be used for FSK only. I suggest we rename that to `maxFskLen` or make `len` a required parameter for all receive/transmit functions, removing the need to record an implicit `len.`

In addition, this `readData` call (which uses len) seems operationally unreachable https://github.com/jgromes/RadioLib/blob/e39c804c73f7308dfbc4f5e988570dca8d8e4690/src/modules/SX126x/SX126x.cpp#L333 -- because `fixImplicitTimeout` would have to fail in order to reach it.

jgromes, 2020-01-10:
> make len a required parameter for all receive/transmit functions

Not sure about that, in explicit header mode, the received packet length is determined automatically upon reception. We also can't easily add a new parameter to receive methods, because of `PhysicalLayer` methods. I think we can just use the existing `len` parameter to pass the expected length. One thing I forgot: make sure `getPacketLength` works correctly in implicit mode.

>In addition, this readData call (which uses len) seems operationally unreachable

If `fixImplicitTimeout` failed, it would then return and never reach `readData`. I think it's correct the way it is now, isn't it?

naggie, 2020-01-10:
> > make len a required parameter for all receive/transmit functions
> 
> Not sure about that, in explicit header mode, the received packet length is determined automatically upon reception. 

Oh yes, of course. 

> We also can't easily add a new parameter to receive methods, because of `PhysicalLayer` methods. I think we can just use the existing `len` parameter to pass the expected length. 

OK

> One thing I forgot: make sure `getPacketLength` works correctly in implicit mode.

Already noted in [my PR](https://github.com/jgromes/RadioLib/pull/103) checklist 👍 

> If `fixImplicitTimeout` failed, it would then return and never reach `readData`. I think it's correct the way it is now, isn't it?

Oops yes I read the comparison wrong. _Gets more coffee_

naggie, 2020-01-10:
OK the PR is ready for review. I've tested it on SX1262 hardware and also updated getTimeOnAir according to the datasheet.

jgromes, 2020-01-13:
Merged #103

closed 2020-01-13 by jgromes