jgromes/RadioLib #87: Errors compiling Examples Transmit and Receive with latest version library on Artemis Redboard

cvoalex, 2019-12-12:
Latest version trying to compile the Transmit and Receive demos for SX126x with an Artemis Redboard. Seems new function setpin that was not in previous versions doesnt have correct parameters when compiled for Artemis.

"C:\Users\Alexander\AppData\Local\Arduino15\packages\SparkFun\tools\arm-none-eabi-gcc\8-2018-q4-major/bin/arm-none-eabi-g++" -DPART_apollo3 -DAM_PACKAGE_BGA -DAM_PART_APOLLO3 -c -g -MMD -mcpu=cortex-m4 -mthumb -mfloat-abi=hard -fdata-sections -Os -ffunction-sections -std=gnu++11 -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DF_CPU=48000000L -DARDUINO=10810 -DARDUINO_AM_AP3_SFE_BB_ARTEMIS -DARDUINO_ARCH_APOLLO3 "-IC:\Users\Alexander\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\1.0.17\variants\SparkFun_RedBoard_Artemis/config" "-IC:\Users\Alexander\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\1.0.17\variants\SparkFun_RedBoard_Artemis/bsp" "-IC:\Users\Alexander\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\1.0.17/cores/arduino/ard_sup" "-IC:\Users\Alexander\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\1.0.17/cores/arduino/ard_sup/ard_supers" "-IC:\Users\Alexander\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\1.0.17\cores\arduino" "-IC:\Users\Alexander\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\1.0.17\variants\SparkFun_RedBoard_Artemis" "-IC:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src" "-IC:\Users\Alexander\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\1.0.17\libraries\SPI\src" "-IC:\Users\Alexander\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\1.0.17\libraries\SoftwareSerial\src" "-IC:\Users\Alexander\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\1.0.17/cores/arduino/am_sdk_ap3/mcu/apollo3" "-IC:\Users\Alexander\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\1.0.17/cores/arduino/am_sdk_ap3/mcu/apollo3/hal" "-IC:\Users\Alexander\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\1.0.17/cores/arduino/am_sdk_ap3/mcu/apollo3/regs" "-IC:\Users\Alexander\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\1.0.17/cores/arduino/am_sdk_ap3/utils" "-IC:\Users\Alexander\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\1.0.17/cores/arduino/am_sdk_ap3/devices" "-IC:\Users\Alexander\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\1.0.17/cores/arduino/am_sdk_ap3/CMSIS/AmbiqMicro/Include" "-IC:\Users\Alexander\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\1.0.17/cores/arduino/am_sdk_ap3/CMSIS/ARM/Include" "C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\ISerial.cpp" -o "C:\Users\ALEXAN~1\AppData\Local\Temp\arduino_build_628036\libraries\RadioLib\ISerial.cpp.o"
C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\Module.cpp: In member function 'void Module::init(uint8_t, uint8_t)':
C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\Module.cpp:59:25: error: no matching function for call to 'Module::setPin(int&, const am_hal_gpio_pincfg_t&)'
setPin(_cs, OUTPUT);
^
In file included from C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\Module.cpp:1:
C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\Module.h:327:10: note: candidate: 'void Module::setPin(int16_t, uint8_t)'
void setPin(int16_t pin, uint8_t mode);
^~~~~~
C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\Module.h:327:10: note: no known conversion for argument 2 from 'const am_hal_gpio_pincfg_t' to 'uint8_t' {aka 'unsigned char'}
C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\Module.cpp:79:26: error: no matching function for call to 'Module::setPin(int&, const am_hal_gpio_pincfg_t&)'
setPin(_int0, INPUT);
^
In file included from C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\Module.cpp:1:
C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\Module.h:327:10: note: candidate: 'void Module::setPin(int16_t, uint8_t)'
void setPin(int16_t pin, uint8_t mode);
^~~~~~
C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\Module.h:327:10: note: no known conversion for argument 2 from 'const am_hal_gpio_pincfg_t' to 'uint8_t' {aka 'unsigned char'}
C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\Module.cpp:82:26: error: no matching function for call to 'Module::setPin(int&, const am_hal_gpio_pincfg_t&)'
setPin(_int1, INPUT);
^
In file included from C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\Module.cpp:1:
C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\Module.h:327:10: note: candidate: 'void Module::setPin(int16_t, uint8_t)'
void setPin(int16_t pin, uint8_t mode);
^~~~~~
C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\Module.h:327:10: note: no known conversion for argument 2 from 'const am_hal_gpio_pincfg_t' to 'uint8_t' {aka 'unsigned char'}
C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\Module.cpp:85:26: error: no matching function for call to 'Module::setPin(int&, const am_hal_gpio_pincfg_t&)'
setPin(_int0, INPUT);
^
In file included from C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\Module.cpp:1:
C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\Module.h:327:10: note: candidate: 'void Module::setPin(int16_t, uint8_t)'
void setPin(int16_t pin, uint8_t mode);
^~~~~~
C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\Module.h:327:10: note: no known conversion for argument 2 from 'const am_hal_gpio_pincfg_t' to 'uint8_t' {aka 'unsigned char'}
C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\Module.cpp:86:26: error: no matching function for call to 'Module::setPin(int&, const am_hal_gpio_pincfg_t&)'
setPin(_int1, INPUT);
^
In file included from C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\Module.cpp:1:
C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\Module.h:327:10: note: candidate: 'void Module::setPin(int16_t, uint8_t)'
void setPin(int16_t pin, uint8_t mode);
^~~~~~
C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\Module.h:327:10: note: no known conversion for argument 2 from 'const am_hal_gpio_pincfg_t' to 'uint8_t' {aka 'unsigned char'}
C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\Module.cpp: In member function 'void Module::setPin(int16_t, uint8_t)':
C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\Module.cpp:258:22: error: no matching function for call to 'pinMode(int16_t&, uint8_t&)'
pinMode(pin, mode);
^
In file included from C:\Users\Alexander\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\1.0.17/cores/arduino/ard_sup/Arduino.h:77,
from C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\TypeDef.h:4,
from C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\Module.h:4,
from C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\Module.cpp:1:
C:\Users\Alexander\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\1.0.17/cores/arduino/ard_sup/ap3_gpio.h:62:6: note: candidate: 'void pinMode(uint8_t, am_hal_gpio_pincfg_t)'
void pinMode(uint8_t pin, am_hal_gpio_pincfg_t mode);
^~~~~~~
C:\Users\Alexander\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\1.0.17/cores/arduino/ard_sup/ap3_gpio.h:62:6: note: no known conversion for argument 2 from 'uint8_t' {aka 'unsigned char'} to 'am_hal_gpio_pincfg_t'
C:\Users\Alexander\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\1.0.17/cores/arduino/ard_sup/ap3_gpio.h:63:6: note: candidate: 'void pinMode(uint8_t, am_hal_gpio_pincfg_t, ap3_err_t*)'
void pinMode(uint8_t pin, am_hal_gpio_pincfg_t mode, ap3_err_t *retval);
^~~~~~~
C:\Users\Alexander\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\1.0.17/cores/arduino/ard_sup/ap3_gpio.h:63:6: note: candidate expects 3 arguments, 2 provided
C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\ISerial.cpp: In member function 'bool ISerial::listen()':
C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\ISerial.cpp:15:38: error: void value not ignored as it ought to be
return(_mod->ModuleSerial->listen());
^
C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\ISerial.cpp: In member function 'bool ISerial::stopListening()':
C:\Users\Alexander\Documents\Arduino\libraries\RadioLib\src\ISerial.cpp:35:45: error: void value not ignored as it ought to be
return(_mod->ModuleSerial->stopListening());

jgromes, 2019-12-12:
Apollo3 is not a platform supported by RadioLib, so if it ever worked at all, it was by random chance rather than by design. I'm also not sure supporting these unicorn MCUs is worthwhile - the Travis CI script already takes half an hour to build all the examples, I don't want to add to it. Though if the build process for Apollo is fast enough, it could be added.

cvoalex, 2019-12-12:
Any chance we can do a consulting fee to get it added?   The new Artemis modules are very awesome.  Fast and small and Sparkfun has some great form factors.

jgromes, 2019-12-12:
I'd rather wait with any changes until the approach in Apollo3 core is decided (https://github.com/sparkfun/Arduino_Apollo3/issues/98).

jgromes, 2019-12-24:
Sparkfun Apollo3 boards should work now, however, they will not be added to the list of platforms that are tested using Travis. The reason is that building all examples for Artemis Redboard takes 45 minutes. This would almost double the time it takes to run a Travis job (see https://travis-ci.org/jgromes/RadioLib/builds/629035240), that's way too much for a board that's not being used by many people.

For that reason, support for Apollo3 will remain unofficial until the build process is significantly optimized.

closed 2019-12-24 by jgromes