jgromes/RadioLib #89: SX1262: Cannot Calibrate image or transmit

naggie, 2019-12-12:
Hi,

I've been trying to successfully transmit a packet using the [SX126x_Transmit](https://github.com/jgromes/RadioLib/blob/master/examples/SX126x/SX126x_Transmit/SX126x_Transmit.ino) example. I have BUSY and DIO1 connected. I've tried both v2.0.1 and master (e61bd58).

The modem I'm using is this: https://www.rfsolutions.co.uk/radio-modules-c10/dip-lora-transceiver-module-20km-868mhz-p807 . It has a regular XTAL, not a TCXO so I don't need to set up DIO3 to setup a TCXO.

# Image calibration
However, whilst the [modem](https://www.rfsolutions.co.uk/radio-modules-c10/dip-lora-transceiver-module-20km-868mhz-p807) successfully initialises many of the parameters, it gets stuck on SetFrequency(868.0) as part of SX1262::Begin() -- specifically when trying to do an Image calibration -- the error status returned is ERR_SPI_CMD_TIMEOUT which corresponds to waiting for the BUSY line. I get the 5 second timeout before failing. I've checked the busy line, it appears to work as expected for the other SPI commands.

# Transmitting
By skipping image calibration (by setting the second arg to SetFrequency to false) I was able to attempt to transmit. When doing this, the code hits the same timeout error but looking at the code it's waiting for int0/DIO1 for the modem to say it's started transmitting.

I am out of ideas as to what the problem is. I've tried using a slower SPI speed, as well as checked connections many times over. I'd greatly appreciate any suggestions.

Thanks for your time.

jgromes, 2019-12-12:
Since your module doesn't have TCXO, did you set the TCXO reference voltage to 0 in the begin method? It's set to 1.6V by default, which will enable TCXO control. Upload the changed code, then unplug the power and wait for a few seconds to make sure everything is completely off. Then plug the power back in.

As far as I know, SX126x doesn't have any documented way to disable TCXO control once it's been enabled, due to the lack of documentation for SPI registers and the fact that the corresponding SPI command can only enable the control, not disable it. Reset or power cycling to make sure all registers have the default value seems to be the only way to do it.

naggie, 2019-12-12:
II literally just tried that before reading your reply after noticing the default of 1.6. --- it worked! Thank you so much!

Callan

closed 2019-12-12 by naggie