jgromes/RadioLib #91: [CC1101] getPacketLength()

Guglio95, 2019-12-17:
Hi Jan,

**Describe the bug**
`getPacketLength(bool update)` says: _"Query modem for the packet length, [if update is true] updates received packet length. Will return cached value when set to false."_

**Expected behavior**
I think the ["&&" should be changed to a "||"](https://github.com/jgromes/RadioLib/blob/86f2d9fd128c935dae27ec4d76c031adff5e6053/src/modules/CC1101/CC1101.cpp#L566): when the driver has just been started `_packetLengthQueried` is `false`, any call to `getPacketLength(false)` will return `0` which is not a cached value. 

Let me know.
Andrea

jgromes, 2019-12-17:
What are the circumstances under which you would call `getPacketLength(false)` after module initialization (i.e. before any packet was received)? I think it's working as intended at the moment - before you received anything, there's no chaced packet length value, so returning 0 in that situation seems acceptable.

Guglio95, 2019-12-17:
Hi Jan,
I'm trying to reduce the traffic on SPI since I'll use multiple peripherals together on the same SPI.

The radio I'm using is set up in _fixed length mode_ so I was trying to get the `packetLen` value once (or sporadically) and reuse in this way:
```cpp
        uint8_t packetLen = getPacketLength(false);
        while (isIncomingDataAvailable()) {
                int16_t result = readData(bytes, packetLen);
                [...]
        }
```

Since the radio is in fixed mode the number of bytes will remain the same until it gets changed via `fixedPacketLengthMode`.

For this reason I was expecting it to retrieve the value if:  
*  `update` is true.
*  if there's no valid cached value (`_packetLengthQueried` is `false`)

For the same reason i suppose [this](https://github.com/jgromes/RadioLib/blob/master/src/modules/CC1101/CC1101.cpp#L283) makes sense only it the radio is in `CC1101_LENGTH_CONFIG_VARIABLE`.

Let me know. 
If you agree with me I'll send a PR. Otherwise, never mind :smile:

jgromes, 2019-12-17:
I think the logic is correct as it is right now - it's there to prevent reading FIFO in variable length mode multiple times.

What you could do is slightly restructure those conditions, since reading CC1101_REG_PKTLEN multiple times in fixed mode doesn't matter.

Guglio95, 2019-12-17:
I see your point about variable length mode.
I'll adapt it as you said;

Thanks :+1:

closed 2019-12-17 by Guglio95