jgromes/RadioLib #92: [CC1101] setModulation method.

Guglio95, 2019-12-17:
Hi Jan,

**Is your feature request related to a problem? Please describe.**
I'm using the CC1101 radio with OOK modulation, in RadioLib it's listed and supported only as a FSK Modem.

**Describe the solution you'd like**
I've already implemented the _basic_ [code](https://github.com/Guglio95/RFQuack/blob/RadioLibSquash/src/radio/RFQCC1101.h#L19) to set it in OOK mode. If you are interested I'll send a pull request to RadioLib. (My implementation is not complete yet, it misses a setOutputPower fix).

**Describe alternatives you've considered**
If so, do you have some tips/preferences? How should I call the method? `setModulation(enum)`

Let me know.

jgromes, 2019-12-17:
There's a similar feature implemented in the SX127x via `int16_t SX127x::setOOK(bool enableOOK)` - I think something similar would be appropriate for the CC1101 driver.

Guglio95, 2019-12-17:
Super, I'll send a PR!

_PS: I fixed the link_

closed 2019-12-20 by jgromes