jgromes/RadioLib #94: Problem initializing the module “DollaTek LORA32 433Mhz ESP32 Sx1278

gmclamo, 2019-12-21:
Hi. I have problems initializing the module “DollaTek LORA32 433Mhz ESP32 Sx1278”. It is a clone of the TGO LoRa32 SX1276 / 78 module. I cannot initialize it with the library <RadioLib.h>. There are problems with the pins. However, I can initialize it with the <Lora.h <library, and I can send and receive packages. The problem is that <Lora.h> does not use pins D0O1 and D0O2, it only uses it like this:

_**//define the pins used by the LoRa transceiver module
#define SCK 5
#define MISO 19
#define MOSI 27
#define SS 18
#define RST 14
#define DIO0 26 
#define BAND 433E6
En el void Loop()
SPI.begin(SCK, MISO, MOSI, SS);
//setup LoRa transceiver module
  LoRa.setPins(SS, RST, DIO0);**_

With this configuration the module works and starts perfectly with lora.h, BUT I NEED TO USE <RadioLib.h>

_// SX1278 has the following connections:
// NSS pin:   18
// DIO0 pin:  26????
// DIO1 pin:  ¿??
SX1278 lora = new Module(18, 26,?);_

With this setting of <RadioLib.h> I cannot initialize the module.

I need help to initialize the module with the <RadioLib.h> library and be able to send and receive packages.

[image]

jgromes, 2019-12-22:
Sorry, at the moment, DIO1 connection is required. There might be an upcoming change in the API where DIO1 requirement will be dropped for SX1278, howeer, I'm not sure when will that happen as I don't have the time to work on it right now.

closed 2019-12-22 by jgromes

jacobeva, 2024-10-03:
@jgromes I would like to help with removing the reliance on the DIO1 pin if possible please, as I have some boards which I need to support which don't have it wired to the board. Should I open a new issue or continue here?

jgromes, 2024-10-03:
@jacobeva actually this is, at least for blocking Rx - you can just pass `RADIOLIB_NC` instead of an actual pin number. However, some features of the SX127x will not be available. The number of functions that can be remapped to DIO0 is limited. CAD, FHSS, direct mode and stream mode still need DIO1.

> Should I open a new issue or continue here?

Preferrably a new one ;)

jacobeva, 2024-10-03:
If that's the case, that would work for me. Would TX work without dio1
as well?

jgromes, 2024-10-03:
@jacobeva

> Would TX work without dio1 as well?

Certainly, that only needs DIO0.

jacobeva, 2024-10-03:
Thank you very much, I'll just use that workaround for now then.