jgromes/RadioLib #95: SX1280 support

SensorsIot, 2019-12-21:
**IMPORTANT: Check the wiki**
Before submitting new issue, please check the [Wiki](https://github.com/jgromes/RadioLib/wiki) and the [API documentation](https://jgromes.github.io/RadioLib/). You might find a solution to your issue there. 

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
The SX1280 is an interesting chip. However, not well supported for the Arduino environment.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
I would like your library support this new chip as you support the SX1262

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

jgromes, 2019-12-22:
Sorry, I currently don't have the time to work on this - I don't even have the module to begin with.

However, I will gladly accept PRs for this purpose.

jgromes, 2020-04-07:
Basic SX128x support was added in 55ad72e0e0f9e92440899a70f4fd3fb7b0b95cb0 - right now it's supported on a level similar to SX126x. The following features are still missing:

* FLRC modem
* BLE modem
* long interleaving for LoRa
* multiple sync word checking
* ranging
* advanced ranging

I'll implement them if anyone has a specific use case, but it's not a priority right now.

closed 2020-04-07 by jgromes