jgromes/RadioLib #97: Help with Lora module SPI CMD

vasav123, 2019-12-26:
Hey, I'm having trouble setting up the SPI transaction in working properly. My current setup is using a STM32F103RB with the SX1262MB2xA5 board as an attachment. I modified the example transmit and receive code to change the pin definitions to the correct ones. However, I'm getting -707 error codes, which i believe is ERR_SPI_CMD_FAILED. From what I gather this doesn't mean that the wiring is incorrect. Unfortunately, I don't have a scope to test and I was wondering if you had any insight on what I can test?

Here's the modified code. 
```
/*
   RadioLib SX126x Transmit Example

   This example transmits packets using SX1262 LoRa radio module.
   Each packet contains up to 256 bytes of data, in the form of:
    - Arduino String
    - null-terminated char array (C-string)
    - arbitrary binary data (byte array)

   Other modules from SX126x family can also be used.

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>

// SX1262 has the following connections:
// NSS pin:   10
// DIO1 pin:  2
// BUSY pin:  9
#define NSS_PIN   D7
#define DI01_PIN  D5
#define ASEL_PIN  A2
//#define D102_PIN   3
#define BUSY_PIN   D3
#define MOSI_PIN   D11
#define MISO_PIN   D12
#define SCLK_PIN   D13
#define RESET_PIN A0
SPIClass mySpi(MOSI_PIN,MISO_PIN,SCLK_PIN);
SPISettings spiSettings=SPISettings(1600000, MSBFIRST, SPI_MODE0);
SX1262 lora = new Module(NSS_PIN, DI01_PIN, BUSY_PIN, mySpi, spiSettings);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1262 lora = RadioShield.ModuleA;

void setup() {
  pinMode(RESET_PIN,OUTPUT);
  digitalWrite(RESET_PIN, LOW);
  delay(100);
  digitalWrite(RESET_PIN,HIGH);
  Serial.begin(9600);

  // initialize SX1262 with default settings
  Serial.print(F("[SX1262] Initializing ... "));
  // carrier frequency:           434.0 MHz
  // bandwidth:                   125.0 kHz
  // spreading factor:            9
  // coding rate:                 7
  // sync word:                   0x1424 (private network)
  // output power:                14 dBm
  // current limit:               60 mA
  // preamble length:             8 symbols
  // TCXO voltage:                1.6 V (set to 0 to not use TCXO)
  // CRC:                         enabled
//  float freq = 960;
  int state = lora.begin(915.0);
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}

void loop() {
  Serial.print(F("[SX1262] Transmitting packet ... "));

  // you can transmit C-string or Arduino string up to
  // 256 characters long
  // NOTE: transmit() is a blocking method!
  //       See example SX126x_Transmit_Interrupt for details
  //       on non-blocking transmission method.
  int state = lora.transmit("Hello World!");

  // you can also transmit byte array up to 256 bytes long
  /*
    byte byteArr[] = {0x01, 0x23, 0x45, 0x56, 0x78, 0xAB, 0xCD, 0xEF};
    int state = lora.transmit(byteArr, 8);
  */

  if (state == ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F("success!"));

    // print measured data rate
    Serial.print(F("[SX1262] Datarate:\t"));
    Serial.print(lora.getDataRate());
    Serial.println(F(" bps"));

  } else if (state == ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 256 bytes
    Serial.println(F("too long!"));

  } else if (state == ERR_TX_TIMEOUT) {
    // timeout occured while transmitting packet
    Serial.println(F("timeout!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }

  // wait for a second before transmitting again
  delay(1000);
}
```

Here's the output for the transmit example.

```
[SX1262] Initializing ... Wait for BUSY ... 
CMD	80	
DATW	0	A2	

Wait for BUSY ... 
CMD	96	
DATW	1	A2	

Wait for BUSY ... 
CMD	8F	
DATW	0	A2	0	A2	

Wait for BUSY ... 
CMD	8A	
DATW	1	A2	

Wait for BUSY ... 
CMD	93	
DATW	20	A2	

Wait for BUSY ... 
CMD	88	
DATW	3	A2	16	A2	A	A2	0	A2	0	A2	0	A2	0	A2	

Wait for BUSY ... 
CMD	2	
DATW	3	A2	FF	A2	

Wait for BUSY ... 
CMD	8	
DATW	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	

Wait for BUSY ... 
CMD	89	
DATW	7F	A2	

Wait for BUSY ... 
CMD	80	
DATW	0	A2	

Wait for BUSY ... 
CMD	17	
DATR	0	A2	0	0	0	0	

Wait for BUSY ... 
CMD	97	
DATW	0	A2	0	A2	1	A2	40	A2	

Wait for BUSY ... 
CMD	11	
DATR	0	A2	0	1	

Symbol length: 4.10 ms
Wait for BUSY ... 
CMD	8B	
DATW	9	A2	4	A2	3	A2	0	A2	

Wait for BUSY ... 
CMD	11	
DATR	0	A2	0	1	

Symbol length: 4.10 ms
Wait for BUSY ... 
CMD	8B	
DATW	9	A2	4	A2	3	A2	0	A2	

Wait for BUSY ... 
CMD	11	
DATR	0	A2	0	1	

Symbol length: 4.10 ms
Wait for BUSY ... 
CMD	8B	
DATW	9	A2	4	A2	3	A2	0	A2	

Wait for BUSY ... 
CMD	11	
DATR	0	A2	0	1	

Wait for BUSY ... 
CMD	D	7	40	
DATW	14	A2	24	A2	

Wait for BUSY ... 
CMD	D	8	E7	
DATW	18	A2	

Wait for BUSY ... 
CMD	11	
DATR	0	A2	0	1	

Wait for BUSY ... 
CMD	1D	7	36	
DATR	0	A2	0	D	

Wait for BUSY ... 
CMD	D	7	36	
DATW	9	A2	

Wait for BUSY ... 
CMD	8C	
DATW	0	A2	8	A2	0	A2	FF	A2	1	A2	0	A2	

Wait for BUSY ... 
CMD	9D	
DATW	0	A2	

Wait for BUSY ... 
CMD	98	
DATW	E1	A2	E9	A2	

Wait for BUSY ... 
CMD	86	
DATW	39	AA	30	0	0	0	0	0	

-707
failed, code -707

```

jgromes, 2019-12-26:
From the transcript of SPI comands, it looks like it fails at setting RF frequency (command 0x86), or image calibration (command 0x98). Few things to try could be:

1. Different frequency. It's SX1262, and it should support the full 150 - 960 MHz range, but it's something to check.
2. TCXO - could you try different reference voltage? You're using the default 1.6 V.

vasav123, 2019-12-29:
I set the tcxo value to 0, since i don't think the shield im using allows me to set it, and then I was able to get the commands to work with 434 MHz. Thanks for your help!

closed 2019-12-29 by vasav123