jgromes/RadioLib #100: SX1262 setup fails at CalibrateImage

lillefyr, 2019-12-30:
Hardware setup:

Lolin32 -dev, sx1262 chip connected to the pins described below.
// ref: Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/variants/lolin32/pins_arduino.h
// Pin connections LOLIN32-DEV (D32) and SX1262
// 18 SCK
// 19 MISO
// 27 MOSI
// 5  SS
// RESET 26
#define DIO1 27
#define BUSY 14

**To Reproduce**
```
#include <RadioLib.h>
SX1262 lora = new Module(5, 27, 14, 26);  // nss, DIO1, rst, busy

void setup() {
  Serial.begin(115200);

  int state = lora.begin();
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}

void loop() { delay(1000); }
```
Enable debug in libraries/RadioLib/src/TypeDef.h
```
#define RADIOLIB_DEBUG
#define RADIOLIB_VERBOSE
```
**Expected behavior**
Chip should be initialized and able to receive data.

**Screenshots**
Kindoff:
```
SX1262::begin
freq=434.00
bw=125.00
sf=9
cr=7
syncWord=18
currentLimit=60.00
preambleLength=8
tcxoVoltage=1.60
CMD	80	
DATW	0	A2	

CMD	96	
DATW	1	A2	

CMD	8F	
DATW	0	A2	0	A2	

CMD	8A	
DATW	1	A2	

CMD	93	
DATW	20	A2	

CMD	88	
DATW	3	A2	16	A2	A	A2	0	A2	0	A2	0	A2	0	A2	

CMD	2	
DATW	3	A2	FF	A2	

CMD	8	
DATW	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	

CMD	89	
DATW	7F	A2	

CMD	80	
DATW	0	A2	

CMD	17	
DATR	0	A2	0	0	0	0	

CMD	97	
DATW	0	A2	0	A2	1	A2	40	A2	

CMD	11	
DATR	0	A2	0	1	

Symbol length: 4.10 ms
CMD	8B	
DATW	9	A2	4	A2	3	A2	0	A2	

CMD	11	
DATR	0	A2	0	1	

Symbol length: 4.10 ms
CMD	8B	
DATW	9	A2	4	A2	3	A2	0	A2	

CMD	11	
DATR	0	A2	0	1	

Symbol length: 4.10 ms
CMD	8B	
DATW	9	A2	4	A2	3	A2	0	A2	

CMD	11	
DATR	0	A2	0	1	

CMD	D	7	40	
DATW	14	A2	24	A2	

CMD	D	8	E7	
DATW	18	A2	

CMD	11	
DATR	0	A2	0	1	

CMD	1D	7	36	
DATR	0	A2	0	D	

CMD	D	7	36	
DATW	9	A2	

CMD	8C	
DATW	0	A2	8	A2	0	A2	FF	A2	1	A2	0	A2	

CMD	9D	
DATW	1	A2	

CMD	98	
DATW	6B	A2	6F	A2	

setFrequencyRaw
434.00
setRfFrequency
455081984
CMD	86	
DATW	1B	AA	20	0	0	0	0	0	

failed, code -707
```

**Additional info (please complete):**
 - MCU: esp32 lolin32 dev (v 1.0 )
 - Wireless moduleSX1262, ESP32
 - Arduino IDE version  [e.g. 1.8.10]
 - Library version [e.g. 3.0.0]

lillefyr, 2019-12-30:
Additional info:

*** To reproduce ***
Code above was changed, by removing some error reporting and adding the next steps.
```
#include <RadioLib.h>
SX1262 lora = new Module(5, 27, 14, 26);  // nss, DIO1, rst, busy

void setup() {
  Serial.begin(115200);

  int state = lora.begin();

  lora.setDio1Action(setFlag);

  state = lora.startReceive();
}

void loop() { delay(1000); }
```

*** additional lines logged ***
```
setFrequencyRaw
434.00
setRfFrequency
455081984
CMD	86	
DATW	1B	AA	20	0	0	0	0	0	

SX126x_Receive_Interrupt_lolind32
CMD	8	
DATW	2	A2	62	A2	0	A2	2	A2	0	A2	0	A2	0	A2	0	A2	

CMD	8F	
DATW	0	A2	0	A2	

CMD	2	
DATW	3	A2	FF	A2	

CMD	82	
DATW	FF	A2	FF	A2	FF	A2	
```
Stop in SX126X_CMD_SET_RX.

jgromes, 2019-12-30:
Could you try a frequency in different image calibration band, e.g. 915 MHz?

lillefyr, 2019-12-30:
```
#define LORA_CARRIER_FREQUENCY                          915.0f  // MHz
#define LORA_BANDWIDTH                                  125.0f  // kHz dual sideband
#define LORA_SPREADING_FACTOR                           11
#define LORA_CODINGRATE                                  8     // 4/8, Extended Hamming
#define LORA_OUTPUT_POWER                               21     // dBm
#define LORA_CURRENT_LIMIT                              120    // mA
#define SYNC_WORD_6X                                    0xFF   // RadioLib uses uint8_t for SX126x
#define LORA_PREAMBLE_LENGTH                             8     // Same for Tx and Rx

  int state = lora.begin(LORA_CARRIER_FREQUENCY,
                          LORA_BANDWIDTH,
                          LORA_SPREADING_FACTOR,
                          LORA_CODINGRATE,
                          SYNC_WORD_6X,
                          17,
                          (uint8_t)LORA_CURRENT_LIMIT);
```

*** output ***
```
SX1262::begin
freq=915.00
bw=125.00
sf=11
cr=8
syncWord=255
currentLimit=120.00
preambleLength=8
tcxoVoltage=1.60
CMD	80	
DATW	0	A2	

CMD	96	
DATW	1	A2	

CMD	8F	
DATW	0	A2	0	A2	

CMD	8A	
DATW	1	A2	

CMD	93	
DATW	20	A2	

CMD	88	
DATW	3	A2	18	A2	A	A2	0	A2	0	A2	0	A2	0	A2	

CMD	2	
DATW	3	A2	FF	A2	

CMD	8	
DATW	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	

CMD	89	
DATW	7F	A2	

CMD	80	
DATW	0	A2	

CMD	17	
DATR	0	A2	0	0	0	0	

CMD	97	
DATW	0	A2	0	A2	1	A2	40	A2	

CMD	11	
DATR	0	A2	0	1	

Symbol length: 16.38 ms
CMD	8B	
DATW	B	A2	4	A2	3	A2	1	A2	

CMD	11	
DATR	0	A2	0	1	

Symbol length: 16.38 ms
CMD	8B	
DATW	B	A2	4	A2	3	A2	1	A2	

CMD	11	
DATR	0	A2	0	1	

Symbol length: 16.38 ms
CMD	8B	
DATW	B	A2	4	A2	4	A2	1	A2	

CMD	11	
DATR	0	A2	0	1	

CMD	D	7	40	
DATW	F4	A2	F4	A2	

CMD	D	8	E7	
DATW	30	A2	

CMD	11	
DATR	0	A2	0	1	

CMD	1D	7	36	
DATR	0	A2	0	D	

CMD	D	7	36	
DATW	9	A2	

CMD	8C	
DATW	0	A2	8	A2	0	A2	FF	A2	1	A2	0	A2	

CMD	9D	
DATW	1	A2	

CMD	98	
DATW	E1	A2	E9	A2	

setFrequencyRaw
915.00
setRfFrequency
959447040
CMD	86	
DATW	39	AA	30	0	0	0	0	0	

CMD	8	
DATW	2	A8	62	1E	0	FB	2	3F	0	54	0	40	0	FA	0	3F
```

jgromes, 2019-12-30:
Looks like tthe same exact error. Which SX126x module are you using? Does it have TCXO or just a crystal?

lillefyr, 2019-12-30:
This is the link where I bought it:
https://www.aliexpress.com/item/4000366551628.html?spm=a2g0s.9042311.0.0.ed2e4c4d4HBBdm

Module print is LoRa1262/8 V1.1

jgromes, 2019-12-30:
I have never used that module before - I'd suggest trying all the supported TCXO reference voltages, including 0 (to disable TCXO).

lillefyr, 2019-12-30:
Success!!!!

[SX1262] Received packet!
[SX1262] Data:		Hello World!
Details in next comment.

lillefyr, 2019-12-30:
When using the lora.begin(); command on SX126x the following defaults are used:
```
float freq = 434.0
float bw = 125.0
uint8_t sf = 9
uint8_t cr = 7
uint8_t syncWord = SX126X_SYNC_WORD_PRIVATE
nt8_t power = 14
float currentLimit = 60.0
uint16_t preambleLength = 8
float tcxoVoltage = 1.6
```

Depending on the module, there can be a tcxo (Temperature Controlled Oscillator) installed, to help reduce the bandwidth of the module.

More details:  [https://www.loratracker.uk/sx1262-improved-lora-device/](https://www.loratracker.uk/sx1262-improved-lora-device/)

The module I use do not have a tcxo, however the default voltage for the tcxo is set to 1.6 V.

This results in a 'code -707' error:  ERR_SPI_CMD_FAILED.

To fix call lora.begin with specific parameters. for example:
```
#define LORA_CARRIER_FREQUENCY                          434.0f  // 413.9f  // MHz
#define LORA_BANDWIDTH                                  125.0f  // kHz dual sideband
#define LORA_SPREADING_FACTOR                           11
#define LORA_CODINGRATE                                  8     // 4/8, Extended Hamming
#define LORA_OUTPUT_POWER                               21     // dBm
#define LORA_CURRENT_LIMIT                              120    // mA
#define SYNC_WORD_6X                                    0x12   // RadioLib uses uint8_t for SX126x
#define LORA_PREAMBLE_LENGTH                             8     // Same for Tx and Rx
#define TCXO_VOLTAGE                                    0.0    // allowed voltage
  int state = lora.begin(LORA_CARRIER_FREQUENCY,
                          LORA_BANDWIDTH,
                          LORA_SPREADING_FACTOR,
                          LORA_CODINGRATE,
                          SYNC_WORD_6X,
                          17,
                          (uint8_t)LORA_CURRENT_LIMIT,
                          LORA_PREAMBLE_LENGTH,
                          TCXO_VOLTAGE);
```

jgromes, 2019-12-30:
I suspected as much - unfortunately, there is no way to tell from software whether the module contains TCXO or XTAL. As a matter of fact, it's not even possible to distinguish SX1268 from SX1262 ...

closed 2019-12-30 by jgromes