jgromes/RadioLib #99: SX1268 - Receive - Error Stae - 707

sdey76, 2019-12-30:
Hii
Updated the Radio Lib on 29 Dec 2019.
Thanks for your support! Am getting ERROR State 707 with LORA SX1268 - With DORJI Development Board with arduino UNO only on the Receive Example
Same error, when i tried the same with SX1268 and ESP32 WROOM Module

The error is inconsistent, on reflahsing the code, i Get Receive Packed Once then Waiting Message...
One again restarting, the error code is -707, Both on UNO as well as ESP32

Transmit Code Works Fine. Kindly help when possible.
Once again Thanks a lot

jgromes, 2019-12-30:
Which exact Dorji module are you using? Does it have TCXO or just plain crystal? And what's the wiring?

sdey76, 2019-12-30:
Thanks for your reply.
Am using DORJI DFR 1268 T
1. Using DORJI Development board for Arduino UNO
The wiring is as follows  with Arduino UNO - DORJI DEVELP BOARD
http://www.dorji.com/products-detail.php?ProId=63 with SXI268
SX1268 DRFT 1268T http://www.dorji.com/products-detail.php?ProId=64

BUSY = 3
DI0 1 = 5
NSS =7
NRESET = A0
DI02 = 5
MOSI = 11,
MISO =12
SCK = 13
SW = 8
----------------------------------------------
WITH ESP32 WROOM - MODULE
RESET =4
DI01=21
BUSY=22
NSS = 5
SCLK = 18
MISO =19
MOSI = 23
Am not sure about TCXO Sir, right now, will check . Thanks a lot

jgromes, 2019-12-30:
DRF1268T has a builtin-in TCXO, so it requires some reference voltage to be set. The default begin() method takes care of that, so if you used unmodified example code it should work.

I find it strange that only receive example is returning this error code: -707 is related to SPI commands. Could you please enable debug mode by uncommenting the following lines:

https://github.com/jgromes/RadioLib/blob/02ac97b07ef4910aa41c162236bcf0fcb569aaff/src/TypeDef.h#L27-L28

Then upload the receive example and psot the serial monitor output.

sdey76, 2019-12-30:
Dear Sir,

Please find the DEBUG Screen data:
SX1262] Initializing ... Wait for BUSY ... 
CMD	80	
DATW	0	AA	

failed, code -707
[SX1262] Initializing ... Wait for BUSY ... 
CMD	80	
DATW	0	AA	

failed, code -707
Thanks & Regards

jgromes, 2019-12-30:
Could you check the pin mapping in your sketch corresponds to the hardware? Also, could you check if there's a way to disconnect the NRESET pin?

lillefyr, 2019-12-30:
One extra from lolin32-dev (no onboard lora, rather SX1262 chip)
Sketch below:

17:08:04.138 -> SX1262::begin
17:08:04.138 -> freq=434.00
17:08:04.138 -> bw=125.00
17:08:04.138 -> sf=9
17:08:04.138 -> cr=7
17:08:04.138 -> syncWord=18
17:08:04.138 -> currentLimit=60.00
17:08:04.138 -> preambleLength=8
17:08:04.138 -> tcxoVoltage=1.60
17:08:04.138 -> bw=125.00
17:08:04.138 -> sf=9
17:08:04.138 -> cr=7
17:08:04.138 -> syncWord=18
17:08:04.138 -> currentLimit=60.00
17:08:04.138 -> preambleLength=8
17:08:04.176 -> tcxoVoltage=1.60
17:08:04.318 -> standby 0
17:08:04.318 -> Wait for BUSY ... 
17:08:04.318 -> Irq=14
17:08:04.353 -> Rst=26
17:08:04.353 -> Gpi=27
17:08:04.353 -> Rx =27
17:08:04.353 -> Tx =-1
17:08:04.353 -> CMD	80	
17:08:04.353 -> DATW	0	AA	
17:08:04.353 -> 
17:08:04.353 -> failed, code -707
====== Sketch ======
#include <RadioLib.h>
SX1262 lora = new Module(5, 14, 26, 27);  //Port-Pin Input:  SPI select, Reset, Busy, Interrupt DIO1

void setup() {
  Serial.begin(115200);
  int state = lora.begin();
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}
volatile bool receivedFlag = false;
volatile bool enableInterrupt = true;
void setFlag(void) {
  if(!enableInterrupt) { return; }
  receivedFlag = true;
}
void loop() { delay(1000); } //do nothing

jgromes, 2019-12-30:
@lillefyr could you double-check your pin assignment? The order is `Module(nss, dio1, rst, busy);`, yours seems to be different.

sdey76, 2019-12-30:
DISCONNECTED NRESET - A0 to arduino UNO
RESULT :
[SX1A2	0	0	0	20	

failed, code -703
[SX1262] Initializing ... Wait for BUSY ... 
CMD	80	
DATW	0	A2	

Wait for BUSY ... 
CMD	96	
DATW	1	A2	

Wait for BUSY ... 
CMD	8F	
DATW	0	A2	0	A2	

Wait for BUSY ... 
CMD	8A	
DATW	1	A2	

Wait for BUSY ... 
CMD	93	
DATW	20	A2	

Wait for BUSY ... 
CMD	88	
DATW	3	A2	16	A2	A	A2	0	A2	0	A2	0	A2	0	A2	

Wait for BUSY ... 
CMD	2	
DATW	3	A2	FF	A2	

Wait for BUSY ... 
CMD	8	
DATW	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	

Wait for BUSY ... 
CMD	89	
DATW	7F	A2	

Wait for BUSY ... 
CMD	80	
DATW	0	AA	

Wait for BUSY ... 
CMD	17	
DATR	0	A2	0	0	0	20	

failed, code -703

sdey76, 2019-12-30:
PIN MAPPING
RX EXAMPLE 

// SX1262 has the following connections:
// NSS pin:   10
// DIO1 pin:  2
// NRST pin:  3
// BUSY pin:  9
SX1268 lora = new Module(7,5,A0,3);

sdey76, 2019-12-30:
Sir
The DORJI Develoment board with arduino has following Pin Assigned as per the PDF LInk below
http://dorji.com/docs/app/DAD06-sx1262-development-board-for-Arduino-ST-Nucleo.pdf
Regards

lillefyr, 2019-12-30:
Pin assignment changed:
SX1262 lora = new Module(5, 27, 14, 26);  // nss, DIO1, rst, busy

More communication, in the end however still -707

17:51:03.782 -> SX1262::begin
17:51:03.782 -> freq=434.00
17:51:03.782 -> bw=125.00
17:51:03.782 -> sf=9
17:51:03.782 -> cr=7
17:51:03.782 -> syncWord=18
17:51:03.782 -> currentLimit=60.00
17:51:03.782 -> preambleLength=8
17:51:03.782 -> tcxoVoltage=1.60
17:51:03.782 -> bw=125.00
17:51:03.782 -> sf=9
17:51:03.782 -> cr=7
17:51:03.782 -> syncWord=18
17:51:03.782 -> currentLimit=60.00
17:51:03.782 -> preambleLength=8
17:51:03.816 -> tcxoVoltage=1.60
17:51:03.990 -> standby 0
17:51:03.990 -> CMD	80
17:51:03.990 -> DATW	0	A2
17:51:03.990 ->
17:51:03.990 -> CMD	96
17:51:03.990 -> DATW	1	A2
17:51:03.990 ->
17:51:03.990 -> CMD	8F
17:51:03.990 -> DATW	0	A2	0	A2
17:51:03.990 ->
17:51:03.990 -> CMD	8A
17:51:03.990 -> DATW	1	A2
17:51:03.990 ->
17:51:03.990 -> CMD	93
17:51:04.025 -> DATW	20	A2
17:51:04.025 ->
17:51:04.025 -> CMD	88
17:51:04.025 -> DATW	3	A2	16	A2	A	A2	0	A2	0	A2	0	A2	0	A2
17:51:04.025 ->
17:51:04.025 -> CMD	2
17:51:04.025 -> DATW	3	A2	FF	A2
17:51:04.025 ->
17:51:04.025 -> CMD	8
17:51:04.025 -> DATW	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2
17:51:04.025 ->
17:51:04.025 -> CMD	89
17:51:04.025 -> DATW	7F	A2
17:51:04.063 ->
17:51:04.063 -> standby 0
17:51:04.063 -> CMD	80
17:51:04.063 -> DATW	0	A2
17:51:04.063 ->
17:51:04.063 -> SPIreadCommand 2
17:51:04.063 -> CMD	17
17:51:04.063 -> DATR	0	A2	0	0	0	0
17:51:04.063 ->
17:51:04.063 -> CMD	97
17:51:04.063 -> DATW	0	A2	0	A2	1	A2	40	A2
17:51:04.063 ->
17:51:04.063 -> getPacketType
17:51:04.063 -> SPIreadCommand 2
17:51:04.063 -> CMD	11
17:51:04.063 -> DATR	0	A2	0	1
17:51:04.097 ->
17:51:04.097 -> Symbol length: 4.10 ms
17:51:04.097 -> CMD	8B
17:51:04.097 -> DATW	9	A2	4	A2	3	A2	0	A2
17:51:04.097 ->
17:51:04.097 -> getPacketType
17:51:04.097 -> SPIreadCommand 2
17:51:04.097 -> CMD	11
17:51:04.097 -> DATR	0	A2	0	1
17:51:04.097 ->
17:51:04.097 -> Symbol length: 4.10 ms
17:51:04.097 -> CMD	8B
17:51:04.097 -> DATW	9	A2	4	A2	3	A2	0	A2
17:51:04.097 ->
17:51:04.097 -> getPacketType
17:51:04.097 -> SPIreadCommand 2
17:51:04.132 -> CMD	11
17:51:04.132 -> DATR	0	A2	0	1
17:51:04.132 ->
17:51:04.132 -> Symbol length: 4.10 ms
17:51:04.132 -> CMD	8B
17:51:04.132 -> DATW	9	A2	4	A2	3	A2	0	A2
17:51:04.132 ->
17:51:04.132 -> getPacketType
17:51:04.132 -> SPIreadCommand 2
17:51:04.132 -> CMD	11
17:51:04.132 -> DATR	0	A2	0	1
17:51:04.132 ->
17:51:04.132 -> SPIwriteCommand 1
17:51:04.166 -> CMD	D	7	40
17:51:04.166 -> DATW	14	A2	24	A2
17:51:04.166 ->
17:51:04.166 -> SPIwriteCommand 1
17:51:04.166 -> CMD	D	8	E7
17:51:04.166 -> DATW	18	A2
17:51:04.166 ->
17:51:04.166 -> getPacketType
17:51:04.166 -> SPIreadCommand 2
17:51:04.166 -> CMD	11
17:51:04.166 -> DATR	0	A2	0	1
17:51:04.166 ->
17:51:04.166 -> readRegister
17:51:04.166 -> CMD	1D	7	36
17:51:04.166 -> DATR	0	A2	0	D
17:51:04.166 ->
17:51:04.166 -> SPIwriteCommand 1
17:51:04.203 -> CMD	D	7	36
17:51:04.203 -> DATW	9	A2
17:51:04.203 ->
17:51:04.203 -> CMD	8C
17:51:04.203 -> DATW	0	A2	8	A2	0	A2	FF	A2	1	A2	0	A2
17:51:04.203 ->
17:51:04.203 -> CMD	9D
17:51:04.203 -> DATW	1	A2
17:51:04.203 ->
17:51:04.203 -> CMD	98
17:51:04.203 -> DATW	6B	A2	6F	A2
17:51:04.203 ->
17:51:04.203 -> setFrequencyRaw
17:51:04.203 -> 434.00
17:51:04.203 -> setRfFrequency
17:51:04.203 -> 455081984
17:51:04.238 -> CMD	86
17:51:04.238 -> DATW	1B	AA	20	0	0	0	0	0
17:51:04.238 ->
17:51:04.238 -> failed, code -707

jgromes, 2019-12-30:
@sdey76 I've seen the issue with -707 after reset previously, it's somehow related to NRST ending up in some undefined state. When the NRST pin isn't connected to anything, it generally solves the problem. The error code you're getting now is -703, ERR_INVALID_TCXO_VOLTAGE - you probably changed TCXO voltage value to something unsupported.

@lillefyr yours fails at the CalibrateImage command - please open a new issue, as these two don't seem related, thanks.

sdey76, 2019-12-30:
I had tried to change the TCXO VOltage and see, but i have re edited the same to 1.6 V
THe code with NRESET Removed worked once, but again started below 

[SX1262] Initializing ... Wait for BUSY ... 
CMD	80	
DATW	0	52	

Wait for BUSY ... 
CMD	96	
DATW	1	52	

Wait for BUSY ... 
CMD	8F	
DATW	0	52	0	52	

Wait for BUSY ... 
CMD	8A	
DATW	1	52	

Wait for BUSY ... 
CMD	93	
DATW	20	52	

Wait for BUSY ... 
CMD	88	
DATW	3	52	16	52	A	52	0	52	0	52	0	52	0	52	

Wait for BUSY ... 
CMD	2	
DATW	3	52	FF	52	

Wait for BUSY ... 
CMD	8	
DATW	0	52	0	52	0	52	0	52	0	52	0	52	0	52	0	52	

Wait for BUSY ... 
CMD	89	
DATW	7F	52	

Wait for BUSY ... 
CMD	80	
DATW	0	52	

Wait for BUSY ... 
CMD	17	
DATR	0	52	0	52	0	52	

Wait for BUSY ... 
CMD	97	
DATW	0	52	0	52	1	52	40	52	

Wait for BUSY ... 
CMD	11	
DATR	0	52	0	52	

failed, code -20
 Thanks & Regards

jgromes, 2019-12-30:
-20 is incorrect modem - it seems like the module has FSK modem enabled, despite the fact LoRa modem was set previously. Also, for some reason it looks like it has available data (that's the 0x52 status).

I'm not entirely sure what's going on here - I'd suggest reuploading the sketch a couple of times, as well as unplugging everything and then connecting it back.

sdey76, 2019-12-31:
Dear Sir
SUCCESS! - 
Following done :
1. Reflashed the Program 6 Times and Changed A0 nto Digital 14, NRESET Removed
SX1268 lora = new Module(7,5,14,3);  ( I guess it does not have much sense A0 to 14 )
--------------------------------------------------------------------------------
success!
[SX1262] Data:		Hello World!
[SX1262] RSSI:		Wait for BUSY ... 
CMD	14	
DATR	0	A2	0	AC	0	2A	0	AE	

-87.00 dBm
[SX1262] SNR:		Wait for BUSY ... 
CMD	11	
DATR	0	A2	0	1	

Wait for BUSY ... 

Thanks a lot
Wish you and your family a wonderful 2020!
Regards 
Soumitra Dey

closed 2019-12-31 by jgromes

N6RFM, 2020-07-03:
Hi Jan,  Thanks for all the great work on RadioLib.  Not sure though that the TCXO issue reported by @sdey76 is resolved.  I'm having the same problem using either the Dorji DRF1268T or the GNICE RF sx1262/8 v1.1 chip.

For example, when running Fossasat 1B if th3e TCXO_VOLTAGE is 1.6 error code -707 is received.  If the voltage is set to 0, then initially error code -20 is received.  Upon restart after unplug/plug in again, the error clears and the sketch runs normally.  
[test.txt](https://github.com/jgromes/RadioLib/files/4871384/test.txt)  Attached is the step wise output in debug mode.  Not sure if helpful but the GNICE library code includes 
/*
The TCXO is 32MHz, so the frequency step should be FREQ_STEP = 32e6 / (2^25) Hz
*/
#define FREQ_STEP    0.953674

This code may be found here https://www.dropbox.com/sh/rf9vrag19rsbxk2/AAARjpAIcn9o5Y3NqD7nrgXta?dl=0

Vert Best,  Bob N6RFM

jgromes, 2020-07-04:
@N6RFM Could you try out different TCXO voltages? 
If the module is using TCXO and the voltage is set to 0, it should get stuck at calibration. However, there is no documented way to disable TCXO control once it's been enabled - the only way to do seems to be setting TCXO voltage to 0 and then hard-reset the module. I think that's what you observed, so it seems like the GNICE RF doesn't have a TCXO, only plain XTAL. This seems to be supported by the fact that it has DIO3 pad - DIO3 is used as reference voltage source for the TCXO, so having it routed to an external pad would be odd.

The `FREQ_STEP` macro is the frequency step of SX126x frequency synthesizer when using 32 MHz crystal (either XTAL or TCXO, doesn't matter). It's also defined in RadioLib:

https://github.com/jgromes/RadioLib/blob/9758ad1e444548c2184a94f27b2c382532e8f6f8/src/modules/SX126x/SX126x.h#L10

N6RFM, 2020-07-05:
Hi Jan,  
Thanks for suggestion.  I tried 1.6, 1.7, 1.8, 2.2, 2.4, 2.7, 3.0 and 3.3 V.  None of these settings work with the GNICE SX1262/8 V1.1 chip.  To repeat what @sdey76 had observed, I set up an UNO and Dorji DRF 1268T (on a DAD06 breakout board).  For the Dorji setup, the code works fine.  I can thus confirm @sdey76 results.  Which is always good.

To further investigate, I have removed the shield from a GNICE chip.  It appears that there is in fact a TCXO mounted on the board.  Pictures are here - https://www.dropbox.com/sh/nxro3em3h09j6e4/AAB-S__5HDzFrMcxblZwNh-ya?dl=0

Next, I tried the GNICE demo code (also in the dropbox folder).  This code works.  

What I cannot tell is whether or not the GNICE code is enabling the TCXO or leaving it in the off state (0V).  In the demo code version of SX1262.h, the code for the TCXO is #define  DIO3_1_6V  0x00 and so on.

So, I am left wondering is somehow for the GNICE chip the manner (SPI call?) in which the TCXO is enabled or voltage is changed differs from the Dorji chip.  Or maybe even a setting of 0 using the RadioLib does in fact turn on the TXCO at 1.6V?

Sorry, out of my depth here.  :-)  

A copy of the demo code is also in the Dropbox folder.  Maybe the GNICE demo code will reveal the root cause of the apparent difference in Dorji versus Gnice chip behavior? 

Happy to test further suggestions.  Best and very much appreciate your help and advice.  Bob N6RFM

jgromes, 2020-07-05:
The GNICE demo shows they're using 3.3 V reference (line 698), however, they also have the TCXO setup timeout set to 1562.5 us (line 593) - RadioLib uses 5000 us by default. as far as I can tell, that is the only difference in regards to TCXO.

You could try calling `setTCXO(3.3, 1563);` right after begin().

N6RFM, 2020-07-05:
Sorry Jan.  Not sure where/how to make this setTCXO call.  In the Groundstation.ino file?  found here - https://www.dropbox.com/s/zjztrdidhprgfqr/GroundStation.ino?dl=0

or some other way/place.  Not sure on syntax.
Thanks

b.

jgromes, 2020-07-05:
In the GroundStation file, on line 588 (within the the USE_SX126X block), though it's probably a lot better to use the default RadioLib examples for debugging.

N6RFM, 2020-07-05:
Thanks Jan!  That works.  BTW, the timeout value does not seem to make a difference, so went with radio.setTCXO(3.3); 
Good point on debug code.  Will look at that next, now with improved understanding.  Cheers.  Bob

N6RFM, 2020-07-05:
Just closing the loop here.  A much closer look at the devices on the GNICE SX1262/8 v1.1 reveals an SJK 32.000.  This is a SJK(SHENZHEN CRYSTAL TECH) 7E32000E12UCG, a simple quartz crystal unit.  Not a TCXO.  So, Jan you were correct!  Just a plain XTAL.  Thanks great support and patience as I waded through this.

Timvrakas, 2020-09-22:
I just ran into the hiccup with these same modules: https://www.aliexpress.com/item/4000366551628.html
parts of the solution:
- There is no TCXO, so you need to disable the bias (0.0V)
- The TCXO voltage can't be disables over SPI, the module needs to be reset or power cycled

Is there somewhere this should be documented?
Also, has anyone has luck finding these modules _with_ the TCXO? I plan on using them in cold environments, and I'd like temp compensation. I guess I could always swap the crystals...

jgromes, 2020-09-22:
@Timvrakas I use Dorji's DRF1268T modules (notice the T at the end). Be careful when picking them up though, a lot of Ebay/Aliexpress sellers have a similar design, i.e. blue PCB with shielding - same as yours - so always double-check you're actually buying Dorji stuff.

As you correctly noticed, TCXO control can't be disabled once it's been enabled, you must reset the module in order to set it back to XTAL. This issue has popped up a couple times already, so I think I'm actually going to do the reset automatically, when user requests 0 V TCXO reference. I didn't want to to implement this in the past since it poses a risk of losing configuration when used out of order - but then again, it allows simpler recovery from improperly configured module.

Timvrakas, 2020-09-22:
Damn, I just ordered a board with the NiceRF footprint!
Have you done power output tests (at 50ohm) from the Dorji modules? I got 20dBm from the NiceRF ones reliably.
Unfortunately, I (and likely others) have not broken out the reset line to the module, since it's not required for operation. Keep that in mind when you think about how to implement.

Thanks for all the help!

jgromes, 2020-09-22:
> Have you done power output tests (at 50ohm) from the Dorji modules?

We're using DRF1268T on FOSSASAT-1B and 2, output power measurement was a part of the testing campaign. We got 21.5 dBm out of them (output power set to 22 dBm, current limit at 140 mA, measured current draw 100 - 110 mA).

> Unfortunately, I (and likely others) have not broken out the reset line to the module

Unfortunately that seems to be the only was to reset the module.