jgromes/RadioLib #101: Issue with TING-01 SX1278 module

starallo, 2020-01-05:
**Describe the bug**
The transmit example sketch works just fine with an AI RA-02 module but fails with an error 16 on the TING 01 module

**To Reproduce**
Default transmit sketch with the following change

SPIClass *loraSPI = new SPIClass(HSPI);
SX1278 lora = new Module(15, 26, -1,*loraSPI);

**Additional info (please complete):**
 - MCU: ESP32
 - Wireless module type: SX1278 (Ting 01
 - Arduino IDE version  1.8.10
 - Library version 3.0.0

jgromes, 2020-01-05:
-16 is ERR_SPI_WRITE_FAILED, probably an issue with the hardware - please double-check the wiring.

Also, please enable debug mode (uncomment [these lines](https://github.com/jgromes/RadioLib/blob/b657bc35219d0928dfe585c1c076bca382619e30/src/TypeDef.h#L36)) and post the output.

starallo, 2020-01-06:
Capture the debugs of the working (RA-02) and failing (TIng-01).
A compare of the two, shows the following:

Working:
80	10000000	48	1001000	65	1100101	6C	1101100	6C	1101100	6F	1101111	20	100000	57	1010111	6F	1101111	72	1110010	6C	1101100	64	1100100	21	100001	
1	1	89	10001001	
81	10000001	8B	10001011	
1	1	8B	10001011	
1	1	89	10001001	
92	10010010	FF	11111111	
1	1	89	10001001	
81	10000001	89	10001001	
1	1	89	10001001	
 success!
[SX1278] Datarate:	577.15 bps

Failing:
80	10000000	48	1001000	65	1100101	6C	1101100	6C	1101100	6F	1101111	20	100000	57	1010111	6F	1101111	72	1110010	6C	1101100	64	1100100	21	100001	
1	1	89	10001001	
81	10000001	8B	10001011	
1	1	89	10001001	
1	1	89	10001001	
1	1	89	10001001	
1	1	89	10001001	
failed, code -16

Full capture of the failing:
Ting-01

[SX1278] Initializing ... 42	1000010	12	10010	
1	1	89	10001001	
1	1	89	10001001	
1	1	89	10001001	
81	10000001	89	10001001	
1	1	89	10001001	
39	111001	13	10011	
B9	10111001	13	10011	
39	111001	13	10011	
1	1	89	10001001	
81	10000001	89	10001001	
1	1	89	10001001	
B	1011	2B	101011	
8B	10001011	2B	101011	
B	1011	2B	101011	
1	1	89	10001001	
81	10000001	89	10001001	
1	1	89	10001001	
1	1	89	10001001	
20	100000	0	0	
A0	10100000	0	0	
20	100000	0	0	
21	100001	8	1000	
A1	10100001	8	1000	
21	100001	8	1000	
24	100100	0	0	
A4	10100100	0	0	
24	100100	0	0	
1	1	89	10001001	
1	1	89	10001001	
81	10000001	89	10001001	
1	1	89	10001001	
6	110	6C	1101100	
86	10000110	6C	1101100	
6	110	6C	1101100	
7	111	80	10000000	
87	10000111	80	10000000	
7	111	80	10000000	
8	1000	0	0	
88	10001000	0	0	
8	1000	0	0	
1	1	89	10001001	
1	1	89	10001001	
81	10000001	89	10001001	
1	1	89	10001001	
1D	11101	76	1110110	
9D	10011101	76	1110110	
1D	11101	76	1110110	
26	100110	4	100	
A6	10100110	4	100	
26	100110	4	100	
1	1	89	10001001	
1	1	89	10001001	
81	10000001	89	10001001	
1	1	89	10001001	
1D	11101	76	1110110	
9D	10011101	76	1110110	
1D	11101	76	1110110	
1E	11110	94	10010100	
9E	10011110	94	10010100	
1E	11110	94	10010100	
31	110001	C3	11000011	
B1	10110001	C3	11000011	
31	110001	C3	11000011	
37	110111	A	1010	
B7	10110111	A	1010	
37	110111	A	1010	
26	100110	4	100	
A6	10100110	4	100	
26	100110	4	100	
1	1	89	10001001	
1	1	89	10001001	
81	10000001	89	10001001	
1	1	89	10001001	
1D	11101	76	1110110	
9D	10011101	76	1110110	
1D	11101	76	1110110	
1	1	89	10001001	
81	10000001	89	10001001	
1	1	89	10001001	
9	1001	FF	11111111	
89	10001001	FF	11111111	
9	1001	FF	11111111	
9	1001	FF	11111111	
89	10001001	FF	11111111	
9	1001	FF	11111111	
4D	1001101	84	10000100	
CD	11001101	84	10000100	
4D	1001101	84	10000100	
1	1	89	10001001	
1	1	89	10001001	
81	10000001	89	10001001	
1	1	89	10001001	
26	100110	4	100	
A6	10100110	4	100	
26	100110	4	100	
success!
[SX1278] Transmitting packet ... 1	1	89	10001001	
81	10000001	89	10001001	
1	1	89	10001001	
1	1	89	10001001	
1D	11101	76	1110110	
1E	11110	94	10010100	
20	100000	0	0	
21	100001	8	1000	
1	1	89	10001001	
81	10000001	89	10001001	
1	1	89	10001001	
1	1	89	10001001	
40	1000000	40	1000000	
C0	11000000	40	1000000	
40	1000000	40	1000000	
1	1	89	10001001	
92	10010010	FF	11111111	
22	100010	C	1100	
A2	10100010	C	1100	
22	100010	C	1100	
E	1110	0	0	
8E	10001110	0	0	
E	1110	0	0	
D	1101	C	1100	
8D	10001101	0	0	
D	1101	0	0	
80	10000000	48	1001000	65	1100101	6C	1101100	6C	1101100	6F	1101111	20	100000	57	1010111	6F	1101111	72	1110010	6C	1101100	64	1100100	21	100001	
1	1	89	10001001	
81	10000001	8B	10001011	
1	1	89	10001001	
1	1	89	10001001	
1	1	89	10001001	
1	1	89	10001001	
failed, code -16

starallo, 2020-01-06:
enabling further debugs shows:

[SX1278] Transmitting packet ... 1	1	89	10001001	
81	10000001	89	10001001	
1	1	89	10001001	
1	1	89	10001001	
1D	11101	76	1110110	
1E	11110	94	10010100	
20	100000	0	0	
21	100001	8	1000	
1	1	89	10001001	
81	10000001	89	10001001	
1	1	89	10001001	
1	1	89	10001001	
40	1000000	40	1000000	
C0	11000000	40	1000000	
40	1000000	40	1000000	
1	1	89	10001001	
92	10010010	FF	11111111	
22	100010	C	1100	
A2	10100010	C	1100	
22	100010	C	1100	
E	1110	0	0	
8E	10001110	0	0	
E	1110	0	0	
D	1101	C	1100	
8D	10001101	0	0	
D	1101	0	0	
80	10000000	48	1001000	65	1100101	6C	1101100	6C	1101100	6F	1101111	20	100000	57	1010111	6F	1101111	72	1110010	6C	1101100	64	1100100	21	100001	
1	1	89	10001001	
81	10000001	8B	10001011	
1	1	89	10001001	
1	1	89	10001001	
1	1	89	10001001	
1	1	89	10001001	

address:	0x1
bits:		2 0
value:		0b11
current:	0b10001001
mask:		0b111
new:		0b10001011
read:		0b10001001

jgromes, 2020-01-06:
From the transcript I can see that the module is failing to switch from standby to transmit mode - I'm not entirely sure why that could be happenning. What's the power source? Since it fails right at transmission start, perhaps it has not enough power and resets, could you try with reduced output power?

jgromes, 2020-01-20:
Closed due to inactivity, fell free to reopen later.

closed 2020-01-20 by jgromes