jgromes/RadioLib #102: SPI Write Failing on RFM69HCW

mhelwick54, 2020-01-05:
**Describe the bug**
When testing with RadioLib RF69 Transmit Example, `begin()` fails with error code -16. Running on an Arduino Uno WiFi rev 2 with an Adafruit RFM69HCW module connected using the ICSP pins, along with D8, D3, and D2 as CS, DI0, and Reset. 

The steps the program goes through are below:
1. Call `begin()`
2. Read the version from the chip with `SPIreadRegister(RF69_REG_VERSION)`
3. Successfully reads `0x24` from the module, which is the correct version.
4. Call `config()`
5. Call `setMode(RF69_STANDBY)`
6. `setMode()` returns `ERR_SPI_WRITE_FAILED`

Debug output from `SPIsetRegValue()` below:
```
1	1	0	0	
81	10000001	4	100	
1	1	0	0	

address:	0x1
bits:		4 2
value:		0b100
current:	0b0
mask:		0b11100
new:		0b100
read:		0b0

failed, code -16

```
**To Reproduce**
Using RF69_Transmit example code, with only the changes above. 

**Expected behavior**
Example runs as normal, initializing radio.

**Additional info (please complete):**
 - MCU: Arduino Uno WiFi rev 2
 - Wireless module type: Adafruit RFM69HCW
 - Arduino IDE version: 1.8.10
 - Library version: 3.0.0

jgromes, 2020-01-06:
I was unable to reproduce the issue (though I was testing with Arduino Pro Mini as I don't have the Uno WiFi board). From the SPI transcript, it looks like the radio thinks it's in sleep mode as opposed to standby.

I believe there was a similar issue on SX127x previously, which was caused by the fact that RegOpMode needed longer to update than other SPI registers. The issue was solved extending the interval between register checks to 5 ms:

https://github.com/jgromes/RadioLib/blob/1b22d4c84efd159b5ec0430a2d3dc75c1ee1552b/src/modules/SX127x/SX127x.cpp#L1113-L1115

You could try a similar change on your setup, i.e. add a 5ms interval as an argument to the following:

https://github.com/jgromes/RadioLib/blob/1b22d4c84efd159b5ec0430a2d3dc75c1ee1552b/src/modules/RF69/RF69.cpp#L817-L819

mhelwick54, 2020-01-06:
Thanks for the quick reply.

No luck adding a higher interval, either at 5ms or 10ms. I also tried running it in a loop, checking if the state is `ERR_NONE` each time, which was unsuccessful. 

One note, in case you aren't familiar with the board: the Uno WiFi rev 2 is based on the ATMEGA4809 vs the 328 or 328p of something like the Pro Mini or Uno rev3. It introduced some incompatibilities with existing libraries, so there's a high chance it won't even work with this one as it stands. If you have any other suggestions I'd greatly appreciate it, but what I'm trying to do may not be possible.

jgromes, 2020-01-06:
I think it should work - the library compiles for the board, and it seems the SPI communication is working. You could try to just ignore the error code, see if RF69_REG_OP_MODE is the only register that does this. Also, it might be worth trying without resetting the module, i.e. with rts set to NC or -1.

mhelwick54, 2020-01-06:
Ignoring all the `if(state != ERR_NONE) { return(state) }` does get through the setup process, but it looks like almost every SPI write/set fails. There are a few that succeed, but only because the register they write to is already the value trying to be written (for example, there are several time `0b0` is written, and the `currentValue` in `SPIsetRegValue()` is already `0b0`). Once the setup is complete, the main loop runs, with each transmit attempt also returning error 16. Disabling the reset doesn't change anything either.

jgromes, 2020-01-07:
That's weird, I guess SPI isn't working correctly after all - do you have another SPI device that could be used to check whether there are any issues with the communication? We should verify SPI is working - perhaps there's an issue with SPI pin assignment, as some platforms support multiple alternate pin functions?

jgromes, 2020-01-21:
Closed due to inactivity, feel free to reopen later.

closed 2020-01-21 by jgromes