jgromes/RadioLib #104: A question regarding RFM95 module initialization

vukasinsk, 2020-01-12:
After successfully working with SX1278 434MHz module I encountered some issues with RFM95 (868MHz) modules. I tried to upload and run both examples Receive_Interrupt and Transmit_Interrupt from the SX127X label in the "Examples" section.
I am using an Arduino Pro Mini coupled with RFM95. Pin connections, screenshot of the issue and pin illustritations of both SX1278 and RFM95 are uploaded in the pictures. (Debug mode enabled in typedef.h)

P.S
Jan, I tried lora.begin(868.0) at first. But I just got the code -2 issue (like now) and I figured I wasn't allowed to use begin() without fully providing all arguments. Thank you for your lightning quick e-mail reply and for your time! :)
[image]
[image]
[image]
[image]

jgromes, 2020-01-12:
Looks like the version number read form SPI doesn't match the expected for RFM95 (0x11 instead of 0x12). Are you sure the chip is actually RFM95, as opposed to RFM96?

You can provide just the first few parameters, the rest will use the default.

vukasinsk, 2020-01-12:
Checked it on Aliexpress exactly 10 minutes ago. They say the RFM95s use RFM96 chips. What am I supposed to change and where if that's the case?

jgromes, 2020-01-12:
Just change the type, i.e. `RFM96 lora = new Module(10, 2, 9, 3);`

vukasinsk, 2020-01-12:
Just did. Getting code -12 right now.  
#define | ERR_INVALID_FREQUENCY   -12
  | The supplied frequency value is invalid for this module.

Note: Supplied lora.begin() with 868.0

vukasinsk, 2020-01-12:
Here's a link to the product if that can help anything at all.
https://www.aliexpress.com/item/32844406764.html?spm=a2g0s.9042311.0.0.5c654c4d7XBAie

jgromes, 2020-01-12:
The chip on the radio is RFM96, which doesn't support 868 MHz, only 433 - 437.

vukasinsk, 2020-01-12:
That is a bummer. Soldered 20 868MHz antennas. Thank you once again.

jgromes, 2020-01-12:
Check the real hardware whenever possible before buying ;) Photos on the aliexpress listing are actually pretty good, you can see that the chip has "RF96" written on it.

closed 2020-01-12 by jgromes