jgromes/RadioLib #105: Receive only works with a small delay (ESP8266 & CC1101)

TunableRoom9, 2020-01-19:
I was trying some of the examples provided in the repository and noticed that I could transmit without any problems but when trying to receive the device would crash. I added a small delay to the receive function:
```c++
// wait for sync word
  while(!digitalRead(_mod->getIrq())) { delay(1); };

  // wait for packet end
  while(digitalRead(_mod->getIrq())) { delay(1); };
```
and now it works. Is this normal? Maybe I didn't setup/wire my device properly.

jgromes, 2020-01-19:
I'm assuming you're using ESP8266 as the platform (and not as just a WiFi module).

Both of those loops seem to be missing call to `yield()` which could trigger ESP8266's watchdog. Could you try replacing `delay(1);` with `yield()` and see if that still works?

TunableRoom9, 2020-01-19:
> I'm assuming you're using ESP8266 as the platform (and not as just a WiFi module).
> 
> Both of those loops seem to be missing call to `yield()` which could trigger ESP8266's watchdog. Could you try replacing `delay(1);` with `yield()` and see if that still works?

Sorry, I should have mentioned I used the ESP8266 as the platform.

Yes, I can confirm using ```yield()``` does work too.

jgromes, 2020-01-19:
Thanks, it looks like all of those loops will require calls to yield. There's probably similar loops in other module drivers as well. Will fix & release a patch.

jgromes, 2020-01-19:
Fixed in patch 3.1.1, it should be available in Arduino library manager in 1 - 2 hours. Thanks for reporting!

closed 2020-01-19 by jgromes