jgromes/RadioLib #110: Wrong register in setDataShapingOOK?

sjd227, 2020-02-08:
Why does `setDataShapingOOK` (sx1272.cpp, line 312) use the `SX127X_REG_PA_RAMP` register? Shouldn't it use `SX127X_REG_OP_MODE` instead, since that's where data shaping is set for both FSK and OOK? 

Also could you please explain why you need to call this when enabling OOK? Why not stick with the default assignment (no shaping)?

Thanks!
- Steven

jgromes, 2020-02-08:
I think you might be consfusing modulation type and data shaping - those are different things. Modulation type is the modulation that should be used for the transmission - FSK, OOK or LoRa. This is indeed set in `SX127X_REG_OP_MODE`.

However, in FSK and OOK, you also have the option to specify output filter for the signal, either Gaussian with different bandwidth-time products (in FSK), or low pass with different cutoff frequencies (in OOK). Those are in the register named `SX127X_REG_PA_RAMP` - I suggest consulting the datasheet.

> Also could you please explain why you need to call this when enabling OOK?

You don't have to call `setDataShapingOOK` to use OOK, just as you don't have to call `setDataShaping` to use FSK - if you don't call it, the module will simply keep the previous data shaping configuration.

sjd227, 2020-02-08:
The datasheet shows ModulationShaping (which specifies the gaussian filters for FSK and frequency cutoffs for OOK) set to bits 4-3 of the RegOpMode register though. Plus RegPaRamp uses bits 0-3, 4, and 7-5, so not sure why `setDataShapingOOK` would be assigning bits 4-3 on this. See datasheet snippets attached. 

[image]
[image]

jgromes, 2020-02-08:
Ah, I was looking at the SX1278 datasheet and implementation - that one is correct. You're right, on SX1272 data shaping is in SX127X_REG_OP_MODE.

jgromes, 2020-02-08:
Should be fixed now, thanks for reporting!

closed 2020-02-08 by jgromes